ChefKissInc/NootedRed #160: Can't fetch right hwlibs

hi-august, 2023-09-06:
**To Reproduce**
Steps to reproduce the behavior:
1. I used CI #1023 kext crashed

**Screenshots**

[image]
[image]

**System Information:**
 - macOS Version: macOS10.15
 - CPU: AMD Ryzen 7 4800U

VisualEhrmanntraut, 2023-09-06:
Outdated kext

closed 2023-09-06 by VisualEhrmanntraut

VisualEhrmanntraut, 2023-09-06:
@hi-august That was just a README change

VisualEhrmanntraut, 2023-09-06:
I am talking about your screenshots. That's from before CI#1023. CI#1023 crashing is #159

hi-august, 2023-09-06:
[image]
it(CI #1019) boots sucess,crashed CI#1024

VisualEhrmanntraut, 2023-09-06:
Failure screenshot is cut in half; useless.