ChefKissInc/NootedRed #166: Raven requires SMCLightSesnsor.kext

tejasraman, 2023-09-07:
**Describe the bug**
In order to get a display put on Raven-based APUs (e.g the Ryzen 5 3400G, Ryzen 3 3200G) you need to add the SMCLightSensor kext to your config. This causes a 5-6 second black screen hang on boot, but does eventually work correctly.

This was also mentioned [here](https://elitemacx86.com/threads/how-to-enable-amd-integrated-graphics-apu-on-macos-clover-opencore.1156/post-10371)

This occurred on my 3400G based desktop PC.

**To Reproduce**
- Get a 3200G or 3400G :D
- Install macOS (installer shows) and wait until you see a black screen. If you press Esc you will hear VoiceOver. Installing SMCLightSensor.kext seems to fix this.

**Expected behavior**
You will not need SMCLightSensor.kext

**Screenshots**
N/A - black screen or no black screen.....

**System Information:**
macOS Version: macOS 12.6.5 and 11.7.4
AMD B550, Ryzen 5 3400G, 16GB DDR4-3200, SK Hynix Gold S31 SATA SSD (my secondary drive

VisualEhrmanntraut, 2023-09-08:
That is likely because the kext is made to have backlight control at all times. I have personally not heard of anyone experiencing this issue, but I am not sure. I will look into it.

Also... you don't need their guides for this kext; all the information is available on our site.

VisualEhrmanntraut, 2023-09-08:
@tejasraman Can you try iMac20,1 and no SMCLightSensor

tejasraman, 2023-09-08:
> That is likely because the kext is made to have backlight control at all times. I have personally not heard of anyone experiencing this issue, but I am not sure. I will look into it.
> 
> Also... you don't need their guides for this kext; all the information is available on our site.

I didn't use their guide (I prefer to read official documentation as I personally don't like when I'm told stuff like (I followed a video guide)

tejasraman, 2023-09-08:
> @tejasraman Can you try iMac20,1 and no SMCLightSensor

I will try this tomorrow or Saturday.

VisualEhrmanntraut, 2023-09-08:
Oh, my bad. Thought you were the person in that thread. Alright

tejasraman, 2023-09-08:
> Oh, my bad. Thought you were the person in that thread. Alright

It's fine, I'm sorry

VisualEhrmanntraut, 2023-09-12:
@tejasraman?

tejasraman, 2023-09-12:
> @tejasraman?

I'll do it today, I have been a little busy

Wincohax, 2023-09-13:
I think it's worth mentioning that I own a 3400g and I use an iMac 20,1 SMBIOS because either the iMac Pro and the Mac Pro SMBIOS throws me a black screen.

I will test this issue tomorrow and check if SMCLightSensor fix the black screen (which it shouldn't be the case).

VisualEhrmanntraut, 2023-09-23:
Error: Required information request time-out. Issue stale. Closing.

closed 2023-09-23 by VisualEhrmanntraut