ChefKissInc/NootedRed #167: MSI Laptop: Restart when boot to macOS with Nootedred.kext

hoaug-tran, 2023-09-08:
hi, ChefKissInc, thank you for your contribution to the community.

I have successfully installed macOS Ventura 13.5.1 on my MSI Alpha 15 B5EEK. I thought everything would be easy, but unexpectedly when using Nootedred.kext to enable the APU, I encountered a restart error when booting into macOS. Here is my machine configuration:

```
CPU: AMD Ryzen 7 5800H
GPU: Radeon RX6600M 8GB (cannot use because the device does not support MUX Switch so I disabled it by SSDT-NoHybGfx.aml)
RAM: 16GB (x2 8GB) DDR4 3200MHz
SSD: SN750 500GB
WIFI & BT: RZ608 ( i disable it with a SSDT - because it doesn't support Wifi )
```

As I described my error above. When using Nootedred.kext, my computer will restart (verbose runs for a few segments - I didn't have time to take a photo or look at it) when booting into macOS. The reason I think this is a BUG is because when I don't use Nootedred.kext my machine can boot into macOS normally with no problems.

I really think my bios has something wrong with Nootedred.kext (because my bios has Above 4G Decoding and ResizeBarSupport option lol). I read your guide and found that it is necessary to turn off CSM/Legacy Support/.... but my device does not support CSM/Legacy Support at all (it only supports UEFI - MSI support staff told me me that ). Therefore, the device does not have CSM/Legacy Support and only has UEFI.

I got help in Telegram and a friend told me that maybe CSM/Legacy Support on my device is still enabled (it's just hidden) and said I should try Enable Secureboot without the key). I did that but the result is still the same. My computer still restarts when booting into MacOS with Nootedred.kext (Disable Nootedred.kext and the computer can boot into macOS normally even though Secureboot is enabled).

I tried Nootedred.kext different versions (both old - very old and new but the result is still the same. I can't boot into macOS)

Please correct me if I'm wrong. Thank you very much. Have a nice day..

CodeRuner5235, 2023-09-08:
You can use smokeless UMAF to change hidden bios settings on AMD laptops

hoaug-tran, 2023-09-08:
> You can use smokeless UMAF to change hidden bios settings on AMD laptops

thanks, but what setting i must change?. i check all setting in UMAF but dont have CSM/Legacy or somthing like that

dungnguyen2703, 2023-09-08:
Let's disable AGPU and use WhateverGreen with your GPU (RX6600M). You may need to proof gpu Id. (Reference https://github.com/daliansky/minisforum-HX90G-Hackintoshhttps://github.com/daliansky/minisforum-HX90G-Hackintosh for more detail)

VisualEhrmanntraut, 2023-09-08:
Not possible to fix this as there's no way in I/O Kit as far as I know for grabbing VBIOS from Option ROMs

commit 334dc219 (2023-09-11): Linux, fuck you

Fixes #167
closed 2023-09-08 by VisualEhrmanntraut