ChefKissInc/NootedRed #172: USB Wifi Kext Breaks display

Ragnarok93, 2023-09-13:
**Describe the bug**
After installing 13.5.2 update, installing Chris1111's V17 USB Wifi kext breaks display output on reboot.

**To Reproduce**
Steps to reproduce the behavior:
1. Update to 13.5.2
2. Install "https://github.com/chris1111/Wireless-USB-Big-Sur-Adapter/releases/tag/V17"
3. Reboot
4. No display

**Expected behavior**
On 13.5.1 there was no issue. USB Wifi was working, nothing abnormal.

**System Information:**
 - macOS Version: 13.5.2
 - CPU: Ryzen 5 4600H

**Additional context**
I'm unsure if this is an issue with NootedRed, or if it's cause by the Wifi Kext, so i'm reporting the bug to both repos.

VisualEhrmanntraut, 2023-09-13:
No correlation.

Ragnarok93, 2023-09-13:
Any idea for a possible fix? I'm about to manually remove the kexts and rebuild the kextcache and see if display returns

VisualEhrmanntraut, 2023-09-13:
@Ragnarok93 No. I don’t see how a Wi-Fi kext would break display other than if the code is shit and starts corrupting random things in memory.