ChefKissInc/NootedRed #173: Black screen on Apple

DepsCian, 2023-09-14:
**Describe the bug**
Black screen on Apple when boot on MacOS

**To Reproduce**
Steps to reproduce the behavior:
1. Shutdown Hackintosh
2. Start PC and boot up MacOS
3. Wait few seconds
4. Display is off

**Expected behavior**
My dekstop

**System Information:**
 - macOS Version: 13.5.1
 - CPU: AMD Ryzen 3 2200G

**Additional context**
I used Monterey where everything worked without any problems, then I upgraded to Ventura (through App Store), after that the black screen started appearing. Not always, but reset NVRAM helps. If this is a duplicate of some issue, can you please explain how to fix it, because after reading all the issues, I have not found an answer. I use HDMI. + when everything was normal, my keyboard would turn off for a while, but now it always has this problem and the backlight is always on when I boot on.

ExtremeXT, 2023-09-14:
@DepsCian What SMBIOS are you using?

DepsCian, 2023-09-14:
@ExtremeXT iMac20,1

VisualEhrmanntraut, 2023-09-14:
What are your boot args?
Are you using AmdTscSync? Is its Info.plist configured correctly (IOCPUNumber 0 on Monterey and newer)?
Do you have conflicting kexts like WhateverGreen?
Does a build before the new firmware injection such as CI #1022 work 100%?

DepsCian, 2023-09-14:
> What are your boot args? Are you using AmdTscSync? Is its Info.plist configured correctly (IOCPUNumber 0 on Monterey and newer)? Do you have conflicting kexts like WhateverGreen? Does a build before the new firmware injection such as CI #1022 work 100%?

Boot args: npci=0x3000 alcid=50 maxmem=13000
IOCPUNumber set to 1, now I'm set him to 0.
WhateverGreen turned off.
I use Cl #1023

VisualEhrmanntraut, 2023-09-14:
"maxmem"? The fuck? Also, CI#1023 is out of date. We only support the very latest commit.

DepsCian, 2023-09-14:
> "maxmem"? The fuck? Also, CI#1023 is out of date. We only support the very latest commit.

I need maxmem to limit my RAM from 16gb to 13000mb, because some bank on my second RAM card has a bit sector, and APU uses this bit sector, which provokes system hangs.

VisualEhrmanntraut, 2023-09-14:
Eh?

VisualEhrmanntraut, 2023-09-14:
Either way, that CI run is before a lot of bug fixes. You need to update.

DepsCian, 2023-09-14:
> Eh?

bad sector

VisualEhrmanntraut, 2023-09-14:
(The memory is allocated by the BIOS. Not the driver. So I don't see the correlation)

DepsCian, 2023-09-14:
Ok, so far everything is working, but I wouldn't jump to conclusions, for maybe the problem is still there, just "lucky" this time)

VisualEhrmanntraut, 2023-09-14:
@DepsCian You can change hidden settings using Smokeless-UMAF

DepsCian, 2023-09-14:
I don't know exactly what the problem is, but after a while now I can't boot into the system at all. When booting, the Apple logo appears for 1-2 seconds and then the PC reboots and boots into Recovery Utility. (I didn't even select a boot partition when booting)

DepsCian, 2023-09-14:
> post your crash logs from /Libray/logs/DiagnosticReports
https://privatebin.net/?51e44e636ae00828#5rcYxG99BzoZtRCz5AmGBDqtAcWh8guK9xLrG8MrHh9d
Thats newest file in this folder

VisualEhrmanntraut, 2023-09-14:
@DepsCian Remove AmdTscSync

DepsCian, 2023-09-15:
> try get last system.log after drive load so we can have a chance to see where it hangs (maybe with Hackintool.app or other)

If it has to be done in the system, then alas, I can't even boot into the system with WhateverGreen. Immediately when booting EFI partition of the disk does not appear the choice of partition to boot, but Recovery Mode. There is a terminal there by the way, maybe there you can find the entry point of the problem?

> @DepsCian Remove AmdTscSync

did'nt work

VisualEhrmanntraut, 2023-09-15:
> If it has to be done in the system, then alas, I can't even boot into the system with WhateverGreen

Then it's not a NootedRed issue.

closed 2023-09-15 by VisualEhrmanntraut

DepsCian, 2023-09-15:
> Then it's not a NootedRed issue.

I know, but for some reason after doing all these things, macOS started loading weird. Maybe someone can help.