ChefKissInc/NootedRed #174: Electron apps freeze system, turn red, crash/start

Otus9051, 2023-09-14:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Open any Electron App (consider VSCode)
2. Observe
3. See error

**Expected behavior**
It works without freezing the system.

**Screenshots**
[image]
[image]

**System Information:**
 - macOS 13.5.2
 - CPU: AMD Ryzen 3 5300U
 - VRAM: 512MB (cannot extend because HP Laptop)

**Additional context**
Here's some l logs I managed to capture
[message.txt](https://github.com/ChefKissInc/NootedRed/files/12611726/message.txt)

ExtremeXT, 2023-09-14:
@Otus9051 Increase VRAM with Smokeless-UMAF. Also, what's your laptop model? I also have an HP with a 5300U, just interested.

Otus9051, 2023-09-14:
HP-15s EQ2144AU.
Thanks for the suggestion, will try tomorrow.

Otus9051, 2023-09-14:
After trying the suggestion and setting VRAM to 2GB, everything seems fine. Closing issue as solved.

closed 2023-09-14 by Otus9051