ChefKissInc/NootedRed #175: Gigabyte A520M-DS3H and Ryzen 5 5600G can't boot - Black screen and reboot

xandeale, 2023-09-14:
Hello, I can't reach any GUI, computer crashes and/or reboots during verbose after opencore selection, resulting in a black screen (monitor loses connection). Disabling Nootedred fixes the problem but pc boots without graphics acceleration. Apps like VLC and Safari browser are bugged and unusable without acceleration.

It shows some errors during boot verbose like "couldn't alloc applekeystoretest" , "can't build index" , and a lot of others, but it can boot normally if nootedred enabled is set as "false".

I think it has to do with motherboards that require SSDT-CPUR, as i have seen people succeed in activating their 5600G APUs on motherboards that don't require this SSDT.

**Expected behavior**
I expected to reach any graphical interface, even with some minor glitches, but nothing shows on screen, only the verbose if you enable -v boot-args.

**System Information:**

 - Mac OS Ventura 13.5.2 (same error on Big Sur, i couldn't test Monterey because it shows the error OCB: load image failed - unsupported and i couldn't manage to solve this problem on Monterey)
 - CPU: AMD Ryzen 5 5600G

ExtremeXT, 2023-09-14:
@xandeale What SMBIOS are you on? The keystore "errors" are harmless and can be ignored. the SSDT is also unrelated, but you should not use the prebuilt SSDT-CPUR from Dortania. I suggest you join our Telegram group for receiving better help, the link is on our webpage alongside our AMD Hackintosh guide, which explains how to properly gather the SSDTs.

xandeale, 2023-09-14:
imacpro1,1 , i tried others and it doesn't change anything

ExtremeXT, 2023-09-14:
@xandeale iMacPro1,1 and MacPro6,1 (and perhaps MacPro6,1?) are NOT recommended because those Macs do not have iGPUs. Desktops should use iMac20,1. The Telegram group way of communicating still stands.

xandeale, 2023-09-14:
it crashed and rebooted with iMac20,1 too, i think it is another problem that needs to be fixed

VisualEhrmanntraut, 2023-09-14:
Are you using AmdTscSync?

xandeale, 2023-09-14:
yes, i enabled it with 11 on IOCPUNumber is it right?? but nothing changes

VisualEhrmanntraut, 2023-09-14:
No, that's incorrect for Monterey in newer; it needs to be 0. However, AmdTscSync will cause problems if you don't need it.

dajiba36cm, 2023-09-21:
> 不，这对蒙特雷和更新者来说是不正确的;它需要为 0。但是，如果您不需要AmdTscSync，则会导致问题。

Which should be used for 0 and -1？

VisualEhrmanntraut, 2023-09-21:
@dajiba36cm -1? Who said that? Do you have 255 threads? 🤣

dajiba36cm, 2023-09-21:
> -1？谁说的？你有 255 个线程吗？🤣

The author wrote -1 in the Github introduction

VisualEhrmanntraut, 2023-09-21:
@dajiba36cm Number of threads minus one

xandeale, 2023-09-21:
i wonder if anyone succeeded activating it with this motherboard or chipset lol

dajiba36cm, 2023-09-21:
> 线程数减去 1

Taking 5800H as an example, I have never understood whether to fill in 0 or -1 or 15

xandeale, 2023-09-21:
better to disable amdtscsync then

VisualEhrmanntraut, 2023-09-21:
> computer crashes and/or reboots during verbose after opencore selection
Can you be more precise? How early is it rebooting?

xandeale, 2023-09-21:
boot verbose messages appear the same way as without nootedred , but after a few minutes of messages showing up on screen, right at the moment the apple logo should appear, monitor loses signal and keyboard lights go off and computer reboots

xandeale, 2023-09-24:
so... any news about my bug? i see that title has changed. I tried the new version of nred but didn't work :(
There is a guy i'm talking with doing a hackintosh on Gigabyte B550M aorus elite and R5 5600G and he was able to use hardware acceleration without issues. I tried his EFI and i have the same error...

xandeale, 2023-09-26:
[nred_error_log_sonoma.txt](https://github.com/ChefKissInc/NootedRed/files/12732110/nred_error_log_sonoma.txt)

xandeale, 2023-09-27:
It didn't work @jalavoui , it crashed again and rebooted.

VisualEhrmanntraut, 2023-10-20:
Was it ever working?

xandeale, 2023-10-20:
it never worked for me

VisualEhrmanntraut, 2023-10-20:
Issue mislabeled

xandeale, 2023-10-27:
hi, i took the risk and updated successfully my bios to F17 , now it boots :)

ExtremeXT, 2023-10-27:
Resolved by updating the BIOS.

closed 2023-10-27 by ExtremeXT