ChefKissInc/NootedRed #179: Click 'Wi-Fi' in Sytem Settings Restarts the Mac

SalmonPete, 2023-09-25:
**Describe the bug**
Clicking the WiFi button in System Settings restarts my computer. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'System Settings'
2. Click on 'Wi-Fi'
3. Scroll down to 'Shuts Down'
4. Restarts

**Expected behavior**
Opens the Wi-Fi page, to select networks.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**System Information:**
 - macOS Sonoma 14 (23A344)
 - CPU: AMD Ryzen 7 4700G

**Additional context**
This behavior didn't happen in the earlier Public Beta Sonoma updates.

VisualEhrmanntraut, 2023-09-25:
Intel Wi-Fi?

SalmonPete, 2023-09-25:
> Intel Wi-Fi?

Correct, using latest itlwm kext.

VisualEhrmanntraut, 2023-09-25:
Boot with NRed disabled, reproduce the issue. If you are able to reproduce the issue without NootedRed being enabled, then it's not a bug with NootedRed.

SalmonPete, 2023-09-25:
> Boot with NRed disabled, reproduce the issue. If you are able to reproduce the issue without NootedRed being enabled, then it's not a bug with NootedRed.
> 
> 

Same thing, disabled NRed in plist, rebooted and opened system settingS, clicked Wi-Fi and rebooted. Amazed Sonoma still works, albeit with no acceleration!.