ChefKissInc/NootedRed #180: When I use nootedred, I get stuck

TwilightNougat, 2023-10-01:
**Describe the bug**

If I don't enable nootedred, I can boot the system normally, but when I enable nootedred, it will get stuck on the last line in -v mode.I tried big sur and Monterey with the same result. I don't know how to debug or get the output log

**To Reproduce**
Steps to reproduce the behavior:
1. Install the system without enabling nootedred
2. After booting to the desktop for the first time, enable nootedred
3. Then reboot
4. Seeing the error, stuck at the end of the -v output

**Expected behavior**
In the Windows, i can see that the default graphics output is rx 6400. At first, I thought it was a problem with the discrete graphics card's default output, so I used the SSDT-NoHybGfx patch to add the bios path, but to no avail. I found that the name of my core BIOS is different from other people's. I don't know if this has any impact. If that doesn't solve the problem, I plan to try the rename patch next.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**System Information:**
 - macOS Version: macOS Big Sur 11.7.9 (20G1426) and macOS Monterey 12.6.8(21G725) 
 - CPU: AMD Ryzen 7 5800h
 - IGPU: AMD Radeon(TM) Graphics (BIOS Name: \_SB.PCI0.GP17.VGA)
 - GPU0:AMD Radeon RX 6400 (BIOS Name: \_SB.PCI0.GPP0.SWUS.SWDS.VGA) Close with SSDT-NoHybGfx

**Additional context**
I blocked an XCHI and used the Rename PEGP to GFX0 and Rename GFX0 to IGPU patches (of course I tried not doing this and couldn't enable nootedred)
My boot-args only have -v
here is my SSDT-NoHybGfx.dsl
`
/*
* Disable discrete graphics (c) Bumblebee-Project
* >> https://github.com/Bumblebee-Project/Bumblebee/wiki/ACPI-for-Developers#acpi-and-the-nvidia-card
* ---
* Important: Change all ACPI paths accordingly.
* Credit to Maemo for original idea and midi1996 for the SSDT
*/
DefinitionBlock ("", "SSDT", 2, "DRTNIA", "NoHybGfx", 0x00000000)
{
    External (_SB_.PCI0.GPP0.SWUS.SWDS.VGA._DSM, MethodObj)    // dGPU ACPI Path
    External (_SB_.PCI0.GPP0.SWUS.SWDS.VGA._PS3, MethodObj)    // dGPU ACPI Path

    Device (NHG1)
    {
        Name (_HID, "NHG10000")  // _HID: Hardware ID
        Method (_STA, 0, NotSerialized)  // _STA: Status
        {
            If (_OSI ("Darwin"))
            {
                Return (0x0F)
            }
            Else
            {
                Return (Zero)
            }
        }

        Method (_INI, 0, NotSerialized)  // _INI: Initialize
        {
            If (_OSI ("Darwin"))
            {
                // If conditional methods reference creation successful
                If ((CondRefOf (\_SB.PCI0.GPP0.SWUS.SWDS.VGA._DSM) && CondRefOf (\_SB.PCI0.GPP0.SWUS.SWDS.VGA._PS3)))
                {
                    // Card Off Request
                    \_SB.PCI0.GPP0.SWUS.SWDS.VGA._DSM (ToUUID ("a486d8f8-0bda-471b-a72b-6042a6b5bee0"), 0x0100, 0x1A, Buffer (0x04)
                        {
                             0x01, 0x00, 0x00, 0x03                         
                        })
                    // Card Off
                    \_SB.PCI0.GPP0.SWUS.SWDS.VGA._PS3 ()
                }
            }
            Else
            {
            }
        }
    }
}

`

SalmonPete, 2023-10-01:
Might be a silly question, when do you disable NootedRed and reboot, have you then enabled WhateverGreen for your GPU?. Second silly question, do you have any device properties in your EFI for the GPU?, if so you'd have to remove them for NootedRed to boot.

TwilightNougat, 2023-10-01:
> Remove the Device Properties from Your Plist.

Thank you very much for your reply. But I did not enable any graphics driver except nootedred.kext, nor whatevergreen. I also tried without adding any DP before, but it still didn't work. I followed the tutorials on the website exactly [https://chefkissinc.github.io/nred](https://chefkissinc.github.io/nred) 
By the way, I am using HDMI port and I don’t have a DP video cable, I guess it might have something to do with the video output port? Although it is unlikely, I have tried everything I can now

TwilightNougat, 2023-10-01:
> Remove the Device Properties from Your Plist. I see 'RestrictEvents' in your EFI, add **revblock=media** to your boot args.

Thank you, I will try it now

SalmonPete, 2023-10-01:
Have you asked over at AMD OS X?.

VisualEhrmanntraut, 2023-10-01:
Maybe it would work if you were not using a prebuilt EFI + configurator.

closed 2023-10-01 by VisualEhrmanntraut