ChefKissInc/NootedRed #181: Screen goes black on boot.

Trijal08, 2023-10-01:
**Describe the bug**
My screen goes black after booting. I think the graphics are not working for my CPU. It worked on an [older version of NootedRed](https://github.com/ChefKissInc/NootedRed/actions/runs/5846412583) though. 

**To Reproduce**
Steps to reproduce the behavior:
Just booting into macOS Sonoma. 

**Expected behavior**
I expected the kext to just allow me in to macOS. 

**Screenshots**
Screen completely Black but backlight is on. 

**System Information:**
 - macOS Version: macOS 14 (Sonoma)
 - CPU: AMD Ryzen 3 3700U
 - Manufacturer: Hewlett-Packard (HP)
 - Model: HP Laptop 15-ef0038ca
 - Country: Canada

VisualEhrmanntraut, 2023-10-01:
Too old to pin-point issue

DepsCian, 2023-10-01:
Same. This is not the first time I have gotten this error. In the last issue I was advised to use the latest build, but that helped for a while.

Trijal08, 2023-10-02:
> Too old to pinpoint issue

What do you mean by that. Does that mean I have to use the older version?

hoaug-tran, 2023-10-03:
I had the same problem when using the latest kext. Using older version kext does not work

hoaug-tran, 2023-10-03:
hmm, im using this version: https://github.com/ChefKissInc/NootedRed/actions/runs/6157959609
and its ok. with lastest version i got black screen sometimes ( Not every time it boots, but sometimes it happens ). i will try kext you send me above

VisualEhrmanntraut, 2023-10-03:
@hoaug-tran Oh, so it’s not the fw injection? What SMBIOS are you using

Trijal08, 2023-10-03:
> @hoaug-tran Oh, so it’s not the fw injection? What SMBIOS are you using

Sorry to be nosy, but I am using the MacBookPro16,3 as you said in the website. Is it ok?

hoaug-tran, 2023-10-03:
@Trijal08 its OK

Trijal08, 2023-10-03:
> @Trijal08 its OK

BRO NO WAY, I know how to fix it. Just enable "Keep data such as "Datahub"" in PlatformInfo/Quirks.
[image]
Case closed for me!

Trijal08, 2023-10-03:
> @hoaug-tran Oh, so it’s not the fw injection? What SMBIOS are you using

IT IS NOT WORKIN AGAIN BRO!!! TELL ME WHAT DO I DO! Here, take my fricking [EFI](https://drive.google.com/file/d/1iqsp4rMdRrCxTMzBr6KME_oL5UUutM56/view?usp=sharing).

hoaug-tran, 2023-10-03:
@Trijal08 😂🤦‍♂️  wait Visual check it

Trijal08, 2023-10-03:
> @Trijal08 😂🤦‍♂️ wait Visual check it

WDYM? What did you do to fiix that thing!

Trijal08, 2023-10-03:
@VisualEhrmanntraut PLS HELPO BROU

Trijal08, 2023-10-03:
@VisualEhrmanntraut IT IS NOT WORKIN AGAIN BRO!!! TELL ME WHAT DO I DO! Here, take my fricking [EFI](https://drive.google.com/file/d/1iqsp4rMdRrCxTMzBr6KME_oL5UUutM56/view?usp=sharing).

hoaug-tran, 2023-10-03:
@Trijal08 i told you wait, he needs to be able to check. SO PLEASE WAIT. DONT SPAM. OK ?

Trijal08, 2023-10-03:
> @Trijal08 i told you wait, he needs time to be able to check. SO PLEASE WAIT. DONT SPAM. OK ?

Bro are you admin? I never knew bro sorry.

hoaug-tran, 2023-10-03:
@Trijal08 no. im not bro. i am a user who got the same error as you. Our job is just to wait, be patient. Try using an older kext to see if it's better. Have you tried yet ?

VisualEhrmanntraut, 2023-10-03:
Enough.

VisualEhrmanntraut, 2023-10-04:
Please try https://github.com/ChefKissInc/NootedRed/actions/runs/6409958778 once it builds

VisualEhrmanntraut, 2023-10-05:
Unlocked for providing feedback. Please be descriptive (exact behaviour, logs, etc), else your input will be useless.

DepsCian, 2023-10-06:
Black screen not fixed for me. P.S: I'm using HDMI, R3 2200G, Ventura.

VisualEhrmanntraut, 2023-10-06:
@DepsCian > Please be descriptive (exact behaviour, logs, etc)

DepsCian, 2023-10-06:
@VisualEhrmanntraut  > I'd like to send logs, but I don't know where to look at them.

binxr, 2023-10-06:
There are times when my Sonoma just has a black screen, and the numlock is active, (which was not supposed to be active) and after many restarts it finally turns off the numlock and the image appears.

same issue on Monterey, (I upgrade Monterey to Sonoma.)
using a Vega 11 and a Ryzen 5 3350gPRO.

binxr, 2023-10-08:
@VisualEhrmanntraut you can help me? with this bug?

hoaug-tran, 2023-10-10:
@VisualEhrmanntraut here is my logs, Who knows, maybe something can help. https://drive.google.com/file/d/1BaL4XtGQPdae4h5nux0YpnPQGVL-9UzF/view?usp=drivesdk

VisualEhrmanntraut, 2023-10-20:
Is this still relevant with the latest commit

laris, 2023-10-27:
I use latest build and also have same issue, after boot into login homepage, the screen will be black status but the mouse arrow can be moved freely, but the background is black totally. Thank after wait about 1 minutes the normal login home page will be ready and show in screen.

log can be checked from here: https://github.com/ChefKissInc/NootedRed/issues/186#issuecomment-1783144129

GeantW0rld, 2023-11-10:
get same issue with internal screen on my laptop but when I plug in the hdmi I can get the display on the external montior

VisualEhrmanntraut, 2023-11-11:
Unable to reproduce. Closing due to staleness.

closed 2023-11-11 by VisualEhrmanntraut

epokuser, 2024-09-18:
hi still happens on amd radeon vega 8 laptop based on zen+

epokuser, 2024-09-23:
> > hi still happens on amd radeon vega 8 laptop based on zen+
> 
> Known issues and can be fixed with these methods
> 
>     * Use -NRedDPDelay at boot-args
> 
>     * Use SSDT-PNLF with _UID 17 or 18 on SSDTTime (19 gives blackscreen sometimes)
> 
>     * Inject EDID through OpenCore DeviceProperties

hi I tried NRedDPDelay at boot-args yet it did the same thing...
and the last two methods idk how to do them...

alvindimas05, 2024-09-24:
> hi I tried NRedDPDelay at boot-args yet it did the same thing... and the last two methods idk how to do them...

You can tag me on AMD OSX Discord or ChefKissInc telegram to help you (My username is the same)

RickiThaDragon, 2024-11-24:
Installed macOS Catalina per Dortania Desktop Guide for AMD. The PC Installed is HP Probook Ryzen 7 4700U. Everything works great after Install. So i had previously Built NootedRed.kext AND went ahead and placed it where it goes, snapshoted Plist went to boot up and it finishes boot like it should just have a black screen now.  

So I removed WhateverGreen.kext (Booted) Same Problem
Put it Back in there with Nootedred still same Problem. Imusing Shanee patches instead of algreys the two that your allowed to turn on and off depending upon which you want to use. (Could that be the Problem)?

Any helpful Feedback Or Solutions Would Be great. 
Minimal Kexts Installed 
Lilu, VoodooHDA, Virtual SMC, WhateverGreen, Realtek Ethernet, wIfi card is incompatible SO no wifi kexts and no bluetooth kexts are installed either. XHCI Unsupported, USB InjectAll,  thats it. Thanks.