ChefKissInc/NootedRed #182: (R5 3350GPRO VEGA 11) black screen with an older and new EFI

binxr, 2023-10-03:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**System Information:**
 - macOS Version: [e.g. macOS 11.7.4]
 - CPU: [e.g. AMD Ryzen 5 3600G]

**Additional context**
Add any other context about the problem here.

binxr, 2023-10-03:
im trying different "solutions" here but nothing is solving

Trijal08, 2023-10-03:
> im trying different "solutions" here but nothing is solving @VisualEhrmanntraut

Bro I had the same problem with different CPU but I fixed it. Just give me your EFI OK?

VisualEhrmanntraut, 2023-10-04:
Duplicate of #181

closed 2023-10-04 by VisualEhrmanntraut