ChefKissInc/NootedRed #184: Stuck on IOKit Daemon (kernelmanagerd) stall 'IGPU'

nisel11, 2023-10-05:
IOKit Daemon (kernelmanagerd) stall 'IGPU' error on system startup after installation

**To Reproduce**
Steps to reproduce the behavior:
I don't know, maybe it's just my bad luck or something.
Or just 'install Mac OS on my laptop'.

**Expected behavior**
I was expecting macOS to start up, not this error.

**Screenshots**
[image]

**System Information:**
 - macOS Version: Big Sur 11.7.9 Ventura 13.5 Sonoma 14.0
 - CPU: Ryzen 5 3550H

**Additional context**
I checked macOS Big Sur 11.7.9 Ventura 13.5 Sonoma 14.0.
everywhere has it, somehow I managed to use Big Sur 11.7.9 once but after upgrading to Sonoma 14.0 the error reappeared, I'm not sure if I still have the old EFI where it didn't exist but with everything I have the error is there.

I'm still trying to rebuild the old EFI but it looks like I just got very lucky that I was able to get through the initial install and after that the error didn't show up, maybe the problem isn't NootedRed but it still seems IGPU related. It's also very likely that this error only appears before startup, because I know for a fact that after startup I didn't get this error even with EFI from Github which now gives me a negative result

VisualEhrmanntraut, 2023-10-05:
Bug with Lilu. See Acidanthera/bugtracker#2286. 5 months old issue. 5. Months. Old.

closed 2023-10-05 by VisualEhrmanntraut

nisel11, 2023-10-05:
> Bug with Lilu. See [acidanthera/bugtracker#2286](https://github.com/acidanthera/bugtracker/issues/2286). 5 months old issue. 5. Months. Old.

Sorry, I'm pretty dumb, but is there any way to boot to the startup setup?

VisualEhrmanntraut, 2023-10-05:
@nisel11 Disable NootedRed temporarily to pass that stage, then reenable it afterwards

mame-erl, 2024-02-21:
> @nisel11 Disable NootedRed temporarily to pass that stage, then reenable it afterwards

Thank you so much, that did the trick for me as well. I was experiencing the same issue on Ryzen 5700U MiniPC.