ChefKissInc/NootedRed #186: Kernel panic when changing wallpaper.

MarkoPrugic, 2023-10-11:
**Describe the bug**
Upon opening settings in wallpaper section, system will hang, also setting wallpaper to any other than static image will cause system to lock up.

**To Reproduce**
Steps to reproduce the behavior:
1. Open System Settings
2. Click on 'Wallpaper'

**Expected behavior**
Working wallpaper selection.

**System Information:**
 - macOS Version: MacOS Sonoma 14.0 (23A344)
 - CPU: AMD Ryzen 3 2200G
 - GPU: Vega 8 (2GB)
 - MB: Asus A320MK
 - RAM: 16GB
 - SMBIOS: iMacPro1,1

**Additional context**
Kernel panic details after reboot from hang:

Panic(CPU 0, time 814371219074): NMIPI for unresponsive processor: TLB flush timeout, TLB state:0x0
RAX: 0xffffffb6549ed000, RBX: 0xfffffffff3b11000, RCX: 0x000000000005a1cc, RDX: 0xffffffffef381000
RSP: 0xffffff8092d8fd30, RBP: 0xffffff8092d8fd40, RSI: 0x0000000000016873, RDI: 0xfffffffff3b11000
R8:  0x0000000000000060, R9:  0x0000000000000001, R10: 0xffffff80292397b0, R11: 0x00000000000e3d9b
R12: 0x0000000000000002, R13: 0x000000000000000e, R14: 0x00000000ffffffff, R15: 0xfffffffff3b11000
RFL: 0x0000000000000287, RIP: 0xffffff7f9e46f9d9, CS:  0x0000000000000008, SS:  0x0000000000000010
Panicked task 0xffffff97a6d50848: 217 threads: pid 0: kernel_task
Backtrace (CPU 0), panicked thread: 0xffffffa1405e25a8, Frame : Return Address
0xffffffac7268af80 : 0xffffff8019986ad6 mach_kernel : _interrupt + 0xd6
0xffffffac7268afe0 : 0xffffff80197d6a7a mach_kernel : _hndl_allintrs + 0x9a
0xffffff8092d8fd40 : 0xffffff7f9e48ac63 com.apple.kext.AMDRadeonX5000 : __ZN32AMDRadeonX5000_AMDVega10Hardware20writeDiagnosisReportERPcRj + 0x4b
0xffffff8092d8fd70 : 0xffffff7f9e422143 com.apple.kext.AMDRadeonX5000 : __ZN37AMDRadeonX5000_AMDGraphicsAccelerator20writeDiagnosisReportERPcRj + 0x3b7
0xffffff8092d8fdc0 : 0xffffff7f9e4265d8 com.apple.kext.AMDRadeonX5000 : __ZN30AMDRadeonX5000_AMDAccelChannel26getHardwareDiagnosisReportEPj + 0x1e2
0xffffff8092d8fe10 : 0xffffff7fae61f8b6 com.apple.iokit.IOAcceleratorFamily2 : __ZN19IOAccelFIFOChannel27restartEv + 0x10a
0xffffff8092d8fe60 : 0xffffff7fae62dea6 com.apple.iokit.IOAcceleratorFamily2 : __ZN20IOAccelEventMachine215restart_channelEv + 0xf6
0xffffff8092d8fea0 : 0xffffff7fae62d56f com.apple.iokit.IOAcceleratorFamily2 : __ZN20IOAccelEventMachine218hardwareErrorEventEv + 0x1db
0xffffff8092d8fed0 : 0xffffff8019f0ef3a mach_kernel : __ZN22IOInterruptEventSource12checkForWorkEv + 0x12a
0xffffff8092d8ff20 : 0xffffff8019f0d75e mach_kernel : __ZN10IOWorkLoop15runEventSourcesEv + 0x13e
0xffffff8092d8ff60 : 0xffffff8019f0cd86 mach_kernel : __ZN10IOWorkLoop10threadMainEv + 0x36
0xffffff8092d8ffa0 : 0xffffff80197d619e mach_kernel : _call_continuation + 0x2e
      Kernel Extensions in backtrace:
         com.apple.iokit.IOAcceleratorFamily2(481.0.1)[E9A83038-06B1-37D9-A1D5-079A67157081]@0xffffff7fae61a000->0xffffff7fae684fff
            dependency: com.apple.driver.AppleMobileFileIntegrity(1.0.5)[F3B44B47-56A7-35C0-994F-CB0DB5DACE3F]@0xffffff801ae60000->0xffffff801ae96fff
            dependency: com.apple.iokit.IOGraphicsFamily(597)[3A88BE82-365C-34DE-83FC-98636FB114B2]@0xffffff7fae7aa000->0xffffff7fae7d8fff
            dependency: com.apple.iokit.IOPCIFamily(2.9)[65146C99-3F6F-39ED-8446-52A48D7F6830]@0xffffff801bb11000->0xffffff801bb42fff
            dependency: com.apple.iokit.IOReportFamily(47)[89BF44D1-200C-3475-A6F5-1F2FC3C335F2]@0xffffff801bb53000->0xffffff801bb55fff
            dependency: com.apple.iokit.IOSurface(352.0.3)[93D55571-8050-3A8B-9159-B33939D35C99]@0xffffff801bc78000->0xffffff801bca1fff
         com.apple.kext.AMDRadeonX5000(5.0.1)[B7FC47F2-A93A-3245-AAA6-42375A9E9B03]@0xffffff7f9e41a000->0xffffff7f9e606fff
            dependency: com.apple.iokit.IOAcceleratorFamily2(481.0.1)[E9A83038-06B1-37D9-A1D5-079A67157081]@0xffffff7fae61a000->0xffffff7fae684fff
            dependency: com.apple.iokit.IOGraphicsFamily(597)[3A88BE82-365C-34DE-83FC-98636FB114B2]@0xffffff7fae7aa000->0xffffff7fae7d8fff
            dependency: com.apple.iokit.IOPCIFamily(2.9)[65146C99-3F6F-39ED-8446-52A48D7F6830]@0xffffff801bb11000->0xffffff801bb42fff
            dependency: com.apple.iokit.IOSurface(352.0.3)[93D55571-8050-3A8B-9159-B33939D35C99]@0xffffff801bc78000->0xffffff801bca1fff
      Kernel Extensions in backtrace:
         com.apple.kext.AMDRadeonX5000(5.0.1)[B7FC47F2-A93A-3245-AAA6-42375A9E9B03]@0xffffff7f9e41a000->0xffffff7f9e606fff
            dependency: com.apple.iokit.IOAcceleratorFamily2(481.0.1)[E9A83038-06B1-37D9-A1D5-079A67157081]@0xffffff7fae61a000->0xffffff7fae684fff
            dependency: com.apple.iokit.IOGraphicsFamily(597)[3A88BE82-365C-34DE-83FC-98636FB114B2]@0xffffff7fae7aa000->0xffffff7fae7d8fff
            dependency: com.apple.iokit.IOPCIFamily(2.9)[65146C99-3F6F-39ED-8446-52A48D7F6830]@0xffffff801bb11000->0xffffff801bb42fff
            dependency: com.apple.iokit.IOSurface(352.0.3)[93D55571-8050-3A8B-9159-B33939D35C99]@0xffffff801bc78000->0xffffff801bca1fff

Process name corresponding to current thread (0xffffffa1405e25a8): kernel_task
Boot args: -v keepsyms=1 debug=0x100 alcid=1 -radcodec unfairgva=1 -wegnoegpu 

Mac OS version:
23A344

Kernel version:
Darwin Kernel Version 23.0.0: Fri Sep 15 14:42:42 PDT 2023; root:xnu-10002.1.13~1/RELEASE_X86_64
Kernel UUID: 0905511D-D697-32D8-8069-EA653FED4635
roots installed: 0
KernelCache slide: 0x0000000019400000
KernelCache base:  0xffffff8019600000
Kernel slide:      0x00000000194e0000
Kernel text base:  0xffffff80196e0000
__HIB  text base: 0xffffff8019500000
System model name: iMacPro1,1 (Mac-7BA5B2D9E42DDD94)
System shutdown begun: NO
Panic diags file available: YES (0x0)
Hibernation exit count: 0

System uptime in nanoseconds: 814371603272
Last Sleep:           absolute           base_tsc          base_nano
  Uptime  : 0x000000bd9c54875a
  Sleep   : 0x0000000000000000 0x0000000000000000 0x0000000000000000
  Wake    : 0x0000000000000000 0x00000017f1340027 0x0000000000000000
Compressor Info: 0% of compressed pages limit (OK) and 0% of segments limit (OK) with 0 swapfiles and OK swap space
Zone info:
  Zone map: 0xffffff8c725ce000 - 0xffffffac725ce000
  . PGZ   : 0xffffff8c725ce000 - 0xffffff8c745cf000
  . VM    : 0xffffff8c745cf000 - 0xffffff9140dce000
  . RO    : 0xffffff9140dce000 - 0xffffff92da5ce000
  . GEN0  : 0xffffff92da5ce000 - 0xffffff97a6dce000
  . GEN1  : 0xffffff97a6dce000 - 0xffffff9c735ce000
  . GEN2  : 0xffffff9c735ce000 - 0xffffffa13fdce000
  . GEN3  : 0xffffffa13fdce000 - 0xffffffa60c5ce000
  . DATA  : 0xffffffa60c5ce000 - 0xffffffac725ce000
  Metadata: 0xffffffe5c7069000 - 0xffffffe5e7069000
  Bitmaps : 0xffffffe5e7069000 - 0xffffffe5ea069000
  Extra   : 0 - 0

panic(cpu 1 caller 0xffffff8019961922): IPI timeout, unresponsive CPU bitmap: 0x1, NMIPI acks: 0x0, now: 0x1, deadline: 814117642118, pre-NMIPI time: 0xbd8d3160be, current: 0xbd9c56e880, global: 0 @pmap.c:3092
Panicked task 0xffffff97a6d50848: 217 threads: pid 0: kernel_task
Backtrace (CPU 1), panicked thread: 0xffffffa13ff05b50, Frame : Return Address
0xffffff8092f87800 : 0xffffff801983a111 mach_kernel : _handle_debugger_trap + 0x4b1
0xffffff8092f87850 : 0xffffff8019997be0 mach_kernel : _kdp_i386_trap + 0x110
0xffffff8092f87890 : 0xffffff801998732c mach_kernel : _kernel_trap + 0x55c
0xffffff8092f87910 : 0xffffff80197d6951 mach_kernel : _return_from_trap + 0xc1
0xffffff8092f87930 : 0xffffff801983a3fd mach_kernel : _DebuggerTrapWithState + 0x5d
0xffffff8092f87a20 : 0xffffff8019839aa3 mach_kernel : _panic_trap_to_debugger + 0x1e3
0xffffff8092f87a80 : 0xffffff8019fd900b mach_kernel : _panic + 0x84
0xffffff8092f87b70 : 0xffffff8019961922 mach_kernel : _pmap_flush_tlbs + 0x852
0xffffff8092f87c40 : 0xffffff80199682dd mach_kernel : _pmap_remove_range + 0x25d
0xffffff8092f87d10 : 0xffffff8019968ef1 mach_kernel : _pmap_remove_options + 0x231
0xffffff8092f87d60 : 0xffffff80198e8704 mach_kernel : _kmem_alloc_pageable + 0xd4
0xffffff8092f87db0 : 0xffffff80198aec5a mach_kernel : _workload_config_available + 0x7f9a
0xffffff8092f87e70 : 0xffffff80198adba2 mach_kernel : _workload_config_available + 0x6ee2
0xffffff8092f87ea0 : 0xffffff8019893a28 mach_kernel : _thread_call_delayed_timer + 0x508
0xffffff8092f87ee0 : 0xffffff8019894ab8 mach_kernel : _thread_call_delayed_timer + 0x1598
0xffffff8092f87fa0 : 0xffffff80197d619e mach_kernel : _call_continuation + 0x2e

Process name corresponding to current thread (0xffffffa13ff05b50): kernel_task

VisualEhrmanntraut, 2023-10-20:
Is this still relevant with the latest commit

AxerErnest19, 2023-10-21:
I have a similar problem when entering the desktop. It seems to be an NMIPI problem but I don't know if it is related to Nred since when you deactivate it it works normally but without graphic acceleration obviously

VisualEhrmanntraut, 2023-10-21:
With *latest*?

AxerErnest19, 2023-10-21:
That's how it is with the last one, before it caused more problems with the graphics IO, but that disappeared with the last update except for the NMIPI Kernel Panic

VisualEhrmanntraut, 2023-10-21:
Kexts?

AxerErnest19, 2023-10-21:
[image]

This is in Ventura and works fine, when updating to Sonoma the error occurs

VisualEhrmanntraut, 2023-10-21:
Fetch gpuRestart logs from /Library/Logs/DiagnosticReports

AxerErnest19, 2023-10-21:
I get on it

AxerErnest19, 2023-10-21:
I can't find the gpuRestart log, only these 
[image]

AxerErnest19, 2023-10-21:
WTF? When you put a wallpaper that is not dynamic it does not work in any way, that is, just a solid color. If you change it it crashes 💀

VisualEhrmanntraut, 2023-10-21:
@AxerErnest19 What SMBIOS are you using

AxerErnest19, 2023-10-21:
I am using the SMBIOS of a MacBookPro16.3 as it is a laptop

VisualEhrmanntraut, 2023-10-21:
Ok. Seems fine. What happens if you try screen recording, does it crash then?

AxerErnest19, 2023-10-21:
I have not tried recording the screen, for now the only times it crashes is when using a dynamic wallpaper, when entering the settings in the wallpaper section and in Apple Music. they all give the same kernel panic. because in terms of performance it looks brutal, except for those crashes

VisualEhrmanntraut, 2023-10-21:
yes bro I need to you do that so I can understand what exactly is going wrong with the driver in Sonoma

AxerErnest19, 2023-10-21:
Okay okay, give me a moment to try it

AxerErnest19, 2023-10-21:
Sorry for the delay brother, the university and its things. If it lets me record the screen normally, the problem is basically that when I open one of the things mentioned above, the color wheel appears, after which it turns off abruptly. I'm sending you the kernel panic logs for all the problems I had with configuration and Apple Music.
[Logs.zip](https://github.com/ChefKissInc/NootedRed/files/13061713/Logs.zip)

VisualEhrmanntraut, 2023-10-21:
@ExtremeXT cannot reproduce on 14.0

AxerErnest19, 2023-10-21:
What can't be reproduced? Or do you tell him? forgive my ignorance hahaha

VisualEhrmanntraut, 2023-10-21:
@AxerErnest19 The wallpaper crash, it does not happen for team member ExtremeXT.

AxerErnest19, 2023-10-21:
It's strange, because I have seen that several do not have the problem except in my case, which does crash. Although it is not only the wallpaper, also as I said with other apps. So far I only tried it on Apple Music which gives the same problem and result 

I have a Ryzen 5 3500U (Raven/Piccaso) for what it's worth.

BlkDev9, 2023-10-24:
I think I have the same issue which I reported earlier today in the telegram group, any dynamic wallpaper crashes the system with that exact same KP. (Ryzen 3 3200g Picasso)

laris, 2023-10-27:
reproduced issue.

NootedRed-ci1052-20231021-d8d7cd1.kext

[Archive.zip](https://github.com/ChefKissInc/NootedRed/files/13190976/Archive.zip)

VisualEhrmanntraut, 2023-11-11:
Everyone, please retry with latest commit.

tejasraman, 2023-11-12:
Doesn't happen in the latest artifact in 3400G and 5500U.

VisualEhrmanntraut, 2023-11-19:
Assuming the issue's been resolved for everyone then. If anyone has the issue persist with the latest commit, they shall post a comment.

closed 2023-11-19 by VisualEhrmanntraut

AxerErnest19, 2023-11-19:
Updated to Sonoma 14.1.1 and the problem persists with the latest commit, kernel panic is the same. I don't know if it has anything to do with it but I read that the cause could be IOSurface, I have no idea if it could be. 

My CPU is R5 3500U

laris, 2023-11-20:
> Updated to Sonoma 14.1.1 and the problem persists with the latest commit, kernel panic is the same. I don't know if it has anything to do with it but I read that the cause could be IOSurface, I have no idea if it could be.
> 
> My CPU is R5 3500U

show the log:)
`/Library/Logs/DiagnosticReports`

AxerErnest19, 2023-11-20:
> > Se actualizó a Sonoma 14.1.1 y el problema persiste con la última confirmación, el pánico del kernel es el mismo. No sé si tiene algo que ver pero leí que la causa podría ser IOSurface, no tengo idea si podría ser.
> > Mi CPU es R5 3500U
> 
> muestra el registro :) `/Library/Logs/DiagnosticReports`

The record is the same as the one you gave me previously so it doesn't seem to change anything, 
[Logs (1).zip](https://github.com/ChefKissInc/NootedRed/files/13420110/Logs.1.zip)

Vitor-RT, 2024-01-09:
Hi, I'm having the same problem, the same thing happens, I try to select the wallpaper and macOS suddenly turns off.
System Information:
macOS Version: MacOS Sonoma 14.2.1
CPU: AMD Ryzen 5 3500U
GPU: Vega 8 (2GB)
MB: Lenovo 5B20S42802
RAM: 12GB (2GB is automatically reserved for iGPU)
SMBIOS: MacPro7,1

tejasraman, 2024-01-10:
> Hi, I'm having the same problem, the same thing happens, I try to select the wallpaper and macOS suddenly turns off. System Information: macOS Version: MacOS Sonoma 14.2.1 CPU: AMD Ryzen 5 3500U GPU: Vega 8 (2GB) MB: Lenovo 5B20S42802 RAM: 12GB (2GB is automatically reserved for iGPU) SMBIOS: MacPro7,1 [Kernel-2024-01-09-084714.zip](https://github.com/ChefKissInc/NootedRed/files/13876899/Kernel-2024-01-09-084714.zip)

Why are you using MP7,1 on a laptop?

Vitor-RT, 2024-01-10:
> Why are you using MP7,1 on a laptop?

I was having trouble running macOS and ended up finding this github with some information:
https://github.com/mikigal/ryzen-hackintosh
I already had problems using certain SMBIOS in a VirtualBox installation once so I decided to follow what was there just to at least test if it worked. As it ended up working, I left this one for now because I had other problems in the system and I gave priority to solving those, but I was already thinking about modifying it.
In general, my only problem is the wallpaper with the kext. So much so that now I'm using WhateverGreen to use momentarily to get some things done as it's working reasonably well (however, it doesn't have the same speed and fluidity in NootedRed).
I don't know if it's related, but in one of my tests my USB mouse wasn't working and in this test macOS worked without freezing in the first few seconds, and it lasted longer than normal, but I just opened the wallpaper settings and changed it. had the same problem. I think it's something from AMD Picasso because I saw other people saying they had the same problem on github.

tejasraman, 2024-01-10:
> I was having trouble running macOS and ended up finding this github with some information:
> https://github.com/mikigal/ryzen-hackintosh
> I already had problems using certain SMBIOS in a VirtualBox [...]

I can only test so much with the hardware that I have - my Picasso 3400G doesn't have that issue on Sonoma

Vitor-RT, 2024-01-10:
> I can only test so much with the hardware that I have - my Picasso 3400G doesn't have that issue on Sonoma

Could you send the EFI you use on your notebook, to see if there is any different configuration. This way I could compare it with my config.list and see if there is anything I forgot.

tejasraman, 2024-01-10:
> Could you send the EFI you use on your notebook, to see if there is any different configuration. This way I could compare it with my config.list and see if there is anything I forgot.

My only Ryzen laptop is based on Lucienne (5000u), sorry. My desktop is Picasso (3000U/3000G) w. Zen+ so I don't think my EFI would ever work...

Vitor-RT, 2024-01-10:
I did some tests here and I think the problem is actually in SATA-unsupported.kext, I had it activated, after deactivating NootedRed it was stable, this kext brings more instability to the system. Its best to use NootedRed only with Ryzen-CtlnaAHCIPort.kext.
If the person still wants/needs to use SATA-unsupported.kext, its possible to use WhateverGreen instead, but the performance is much slower than NootedRed.
I only had one problem when opening Brave Browser, because when I opened the browser it started to freeze and there was a Kernel Panic, if anyone want to see the logs:
[Kernel_2024-01-10-022141_Mac-Pro.zip](https://github.com/ChefKissInc/NootedRed/files/13883579/Kernel_2024-01-10-022141_Mac-Pro.zip)
But from what I've seen in other reports here, this is normal.

tejasraman, 2024-01-10:
> I did some tests here and I think the problem is actually in SATA-unsupported.kext, I had it activated, after deactivating NootedRed it was stable, this kext brings more instability to the system. [...]

Aaaah that makes sense. I'm not using SATA unsupported on either the notebook or the desktop :)

VisualEhrmanntraut, 2024-01-10:
SATA-unsupported.kext is for Intel

Nothingelse5545, 2024-01-26:
I am having the same issue in my R5 3500U
Kernel panic while changing the wallpaper, specifically dynamic
Also, in the weather app, the app store and when doing heavy things

Any help will be greatly appreciated

SsNiPeR1, 2024-01-26:
@Nothingelse5545 dynamic wallpapers can cause crashes on ventura+ afaik. the best option is to use image wallpapers, i think

sleepppwalker, 2024-02-26:
I am having the same issue in my r5 3500u.
And if I connect an external monitor, I'll get kernel panic on start.