ChefKissInc/NootedRed #187: Usage of any Browser other than Safari causes hang or hardlock

TheWinHelper, 2023-10-20:
**Describe the bug**
The usage of any browser other than Safari causes issues. Everything based on Firefox causes an imediate hardlock, while Chrome/Vivaldi sometimes work fine and other times lag a lot.

**To Reproduce**
Steps to reproduce the behavior:
Install Firefox
Enjoy System beeing locked up

**Expected behavior**
Browser should not lag that much

**System Information:**
 - macOS Version: macOS 12.6.5
 - CPU: Ryzen 3 5425U with a AMD Radeon RX Vega 6.
 - Device is a Thinkpad E15G4

**Additional context**
When not using NootedRed the browsers seeem to work fine (but then there is no hardware acceleration. Usage of NootedRed also causes issues with waking the device out of sleep for me.

ExtremeXT, 2023-10-20:
@TheWinHelper Duplicate of #158. Sleep should work with the latest commit build.

TheWinHelper, 2023-10-20:
@ExtremeXT 
I use the build from yesterday evening which should be the latest one? Correct?. Sleep Issues are still there. The other Issue is only talking about Chrome based Browsers, but for me even Firefox is not working.

Kivooeo, 2023-10-20:
Confirm.
Ryzen 5 5600H; Apps with HW Acceleration Don't run and freeze system.

VisualEhrmanntraut, 2023-10-20:
Duplicate of #158.

closed 2023-10-20 by VisualEhrmanntraut