ChefKissInc/NootedRed #188: Wake transition timed out after 180 seconds while calling power state change callbacks. Suspected bundle: com.apple.iokit.IOGraphicsFamily. Thread 0x1308a

TwilightNougat, 2023-10-23:
**Describe the bug**
When I wake up from sleep, there is no signal output from the screen and the system automatically restarts after a while.
Here is the log:"Wake transition timed out after 180 seconds while calling power state change callbacks. Suspected bundle: com.apple.iokit.IOGraphicsFamily. Thread 0x1308a"
I used the latest version compiled two days ago, 

**To Reproduce**
Steps to reproduce the behavior:
1. Click the sleep button
2. enters s3 sleep mode (The fan is turned off and the indicator light indicates sleep status)
3. Click any key on the keyboard
4. The fan starts and the indicator light comes on
5. There is no output on the screen. After a while, the computer automatically restarts and outputs a panic log.

**System Information:**
 - macOS Version: MacOS 14.0
 - CPU: AMD Ryzen 7 5800h

**Additional context**
panic(cpu 2 caller 0xffffff8016792667): Wake transition timed out after 180 seconds while calling power state change callbacks. Suspected bundle: com.apple.iokit.IOGraphicsFamily. Thread 0x1308a.
Failure code:: 0x00000800 00000027

Backtracing specified thread
Panicked task 0xffffff9d77eef008: 277 threads: pid 0: kernel_task
Backtrace (CPU 2), panicked thread: 0xffffffa248ff35a8, Frame : Return Address
0xfffffffcdffdf790 : 0xffffff80161818c7 
0xfffffffce047bb70 : 0xffffff8016067336 
0xfffffffce047bc00 : 0xffffff80160654a0 
0xfffffffce047bc50 : 0xffffff801617e800 
0xfffffffce047bc90 : 0xffffff801617dfeb 
0xfffffffce047bcd0 : 0xffffff7faafb401d 
0xfffffffce047bd00 : 0xffffff7faafcdf91 
0xfffffffce047bd10 : 0xffffff7faafb71f8 
0xfffffffce047bda0 : 0xffffff7faafc3c63 
0xfffffffce047bdd0 : 0xffffff80166f29f9 
0xfffffffce047be70 : 0xffffff80166f23e6 
0xfffffffce047bea0 : 0xffffff8016093a28 
0xfffffffce047bee0 : 0xffffff8016094ab8 
0xfffffffce047bfa0 : 0xffffff8015fd619e 
      Kernel Extensions in backtrace:
         com.apple.iokit.IOGraphicsFamily(597.0)[3A88BE82-365C-34DE-83FC-98636FB114B2]@0xffffff7faafaa000->0xffffff7faafd8fff
            dependency: com.apple.iokit.IOPCIFamily(2.9)[65146C99-3F6F-39ED-8446-52A48D7F6830]@0xffffff8018311000->0xffffff8018342fff

Process name corresponding to current thread (0xffffffa248ff35a8): kernel_task

Mac OS version:
23A344

Kernel version:
Darwin Kernel Version 23.0.0: Fri Sep 15 14:42:42 PDT 2023; root:xnu-10002.1.13~1/RELEASE_X86_64
Kernel UUID: 0905511D-D697-32D8-8069-EA653FED4635
roots installed: 0
KernelCache slide: 0x0000000015c00000
KernelCache base:  0xffffff8015e00000
Kernel slide:      0x0000000015ce0000
Kernel text base:  0xffffff8015ee0000
__HIB  text base: 0xffffff8015d00000
System model name: iMac20,1 (Mac-CFF7D910A743CAAF)
System shutdown begun: NO
Panic diags file available: YES (0x0)
Hibernation exit count: 0

System uptime in nanoseconds: 9094284324497
Last Sleep:           absolute           base_tsc          base_nano
  Uptime  : 0x000008456d963195
  Sleep   : 0x00000800fdb0ef59 0x0000000dbee77260 0x0000000000000000
  Wake    : 0x00000801101d112c 0x00000000728f6120 0x000008010fbe1c28
Compressor Info: 0% of compressed pages limit (OK) and 0% of segments limit (OK) with 0 swapfiles and OK swap space
Zone info:
  Zone map: 0xffffff9242bc1000 - 0xffffffb242bc1000
  . PGZ   : 0xffffff9242bc1000 - 0xffffff9245bc2000
  . VM    : 0xffffff9245bc2000 - 0xffffff971215b000
  . RO    : 0xffffff971215b000 - 0xffffff98ab88e000
  . GEN0  : 0xffffff98ab88e000 - 0xffffff9d77e27000
  . GEN1  : 0xffffff9d77e27000 - 0xffffffa2443c0000
  . GEN2  : 0xffffffa2443c0000 - 0xffffffa710959000
  . GEN3  : 0xffffffa710959000 - 0xffffffabdcef3000
  . DATA  : 0xffffffabdcef3000 - 0xffffffb242bc1000
  Metadata: 0xfffffffc9b147000 - 0xfffffffcbb147000
  Bitmaps : 0xfffffffcbb147000 - 0xfffffffcbf947000
  Extra   : 0 - 0

VisualEhrmanntraut, 2023-11-11:
Closing due to staleness.

closed 2023-11-11 by VisualEhrmanntraut

Porco-Rosso, 2023-12-01:
Hi Just want to comment that I have exactly the same issue. Happy to help debug.

Rxxbertx, 2023-12-02:
That cpu needs CPUTSCSYNC BY SEEY6 
https://github.com/Seey6/CpuTscSync

Porco-Rosso, 2023-12-03:
I have a samsung drive installed, but its not the one being used to run MacOS. Will try both potential fixes.