ChefKissInc/NootedRed #189: Enabling nootedred prevents me from entering the system

ShevonKuan, 2023-10-25:
**Describe the bug**

I did not enable nootedred when installing the system. After the installation was completed, I could successfully enter the system.  But when I enabled nootedred I got stuck on the following screen.  The content of this screen keeps looping and cannot be entered into the system. Maybe this is my problem instead of a bug.

**To Reproduce**

After enabling, the problem occurs. After disabling, the problem is solved and you can enter the system.

**Expected behavior**

**Screenshots**

[image]

**System Information:**
 - macOS Version: 13.5.1
 - CPU:  AMD Ryzen 7 4800u
 - wifi: Intel ax210
 - SSD: wd sn750

VisualEhrmanntraut, 2023-10-25:
@ShevonKuan https://chefkissinc.github.io/nred#prerequisites

ShevonKuan, 2023-10-25:
Thank you.  I followed this tutorial and I'm pretty sure I met all the prerequisites.  For example, my computer does not have a discrete graphics card and I have allocated 2g of video memory for the vega8 core display.

VisualEhrmanntraut, 2023-11-11:
Closing as this issue seems to be user error.

closed 2023-11-11 by VisualEhrmanntraut

ryanloee, 2023-12-26:
According to this document, the problem still occurs ：(