ChefKissInc/NootedRed #192: Apps not opening make laptop crash on system when opening them

nanohelperDev, 2023-10-29:
Hello dear,
i face it problem when trying apps like Fl studio and zbrush , its not opening
when i use it fl studio 20 he worked fine but the screen its flickering  but when i upgrade it to version fl studio 21 when i click to open them , he start loading but he make the system crash stuck in loading icon on mouse,

for zbrush he wont open at all , but for me blender and unity 3d working fine also google chrome i disable hardware acceleration on it to make it run without stuck at first, 

my spec its :

Amd rayzen 5625u, Vram 2GB
16Gb ram
HP laptop 15s-eq3XXX

Thanks in advance

VisualEhrmanntraut, 2023-11-11:
Can you retry those apps with the latest commit?

nanohelperDev, 2023-11-12:
Hello My friend , i can confirme its still the same problem , but when i disable nootedred and run with whatevergreen , he work normally but laggy you know.

VisualEhrmanntraut, 2023-11-12:
Using WhateverGreen on an AMD iGPU is the same as not having it, because it does nothing :-)

nanohelperDev, 2023-11-12:
yees sure ;) , also all adobe apps working correcly without any problem exporting and importing everything fine without any third party tool , and the softwares rendering like blender and unity.........etc , works perfectly

nanohelperDev, 2023-11-16:
the new version of Zbrush its working right now , on the last update :)

nanohelperDev, 2023-11-16:
Also for crossover i try it a game on windows , solitair3 and its working fine , and softwares working fine , but games not as you mention before , Thanks

nanohelperDev, 2023-11-18:
Hello new news, i can confirme thats unreal engine 4 / 5 doesn't use graphics acceleration , its laggy sometimes red screen , sometimes get out automatically

VisualEhrmanntraut, 2024-03-13:
Please retry with the latest commit.

szmarczak, 2024-03-26:
I volunteered and:
- FL Studio works like a charm
- Google Chrome launches but is unusuable, if you try to type anything in the address bar the entire screen freezes for like 30 seconds and can't do anything (except the mouse)
- haven't tested other apps yet

VisualEhrmanntraut, 2024-03-26:
Chrome and etc is just #158

VisualEhrmanntraut, 2024-03-31:
Reports seem to suggest this is fixed now, so closing this. Comment if you are still experiencing the issue with latest commit :)

closed 2024-03-31 by VisualEhrmanntraut