ChefKissInc/NootedRed #193: Login screen need wait long time about 30+ secs when boot after version NootedRed-ci1052-20231021-d8d7cd1

laris, 2023-11-01:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
**Each time boot into login screen, the screen will be black background, but mouse can be show and move**

1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.
**After boot to login screen, show login page not black screen**
**Screenshots**
If applicable, add screenshots to help explain your problem.

**System Information:**
 **- macOS Version: [Sonoma v14.0]
 - CPU: [AMD Ryzen 7 5800U]**

**Additional context**
Add any other context about the problem here.
**Latest 3 version tested**

1. NootedRed-ci1052-20231021-d8d7cd1 (ci + number) (build date) (commit id) have issue
2. NootedRed-ci1051-20231021-2c709da have issue
3. NootedRed-ci1049-20231021-18739bc no issue

If need, I can provide more log to check if any logic problems (when control the backlight error method?)

VisualEhrmanntraut, 2023-11-11:
Please try again with latest version.

laris, 2023-11-12:
testing `NootedRed-ci1055-20231111-dfd2ffa-DEBUG`

clean log and restart system, generated some `gpuRestart` log for your ref:

[gpuRestart_Kernel_2023-11-12_clean-restart-before-22h34m.zip](https://github.com/ChefKissInc/NootedRed/files/13328305/gpuRestart_Kernel_2023-11-12_clean-restart-before-22h34m.zip)

laris, 2023-11-12:
```
                ,xNMM.          --------------------
               .OMMMMo           OS: macOS 14.1.1 23B81 x86_64
               OMMM0,            Host: Hackintosh (SMBIOS: MacBookPro16,3)
     .;loddo:' loolloddol;.      Kernel: 23.1.0
   cKMMMMMMMMMMNWMMMMMMMMMM0:    Uptime: 7 mins
 .KMMMMMMMMMMMMMMMMMMMMMMMWd.    Packages: 49 (brew)
 XMMMMMMMMMMMMMMMMMMMMMMMX.      Shell: fish 3.6.1
;MMMMMMMMMMMMMMMMMMMMMMMM:       Resolution: 1440x900
:MMMMMMMMMMMMMMMMMMMMMMMM:       DE: Aqua
.MMMMMMMMMMMMMMMMMMMMMMMMX.      WM: Quartz Compositor
 kMMMMMMMMMMMMMMMMMMMMMMMMWd.    WM Theme: Blue (Dark)
 .XMMMMMMMMMMMMMMMMMMMMMMMMMMk   Terminal: iTerm2
  .XMMMMMMMMMMMMMMMMMMMMMMMMK.   Terminal Font: MonacoNFM 14
    kMMMMMMMMMMMMMMMMMMMMMMd     CPU: AMD Ryzen 7 5800U with Radeon Graphics
     ;KMMMMMMMWXXWMMMMMMMk.      GPU: AMD Radeon Graphics
       .cooc,.    .,coo:.        Memory: 8606MiB / 16384MiB

```

Opencore 0.9.6

laris, 2023-11-12:
upload video to show the behavior about black screen before login.

https://github.com/ChefKissInc/NootedRed/assets/951666/7ccd56bb-0582-4047-87b5-86d2cf259fdd

VisualEhrmanntraut, 2023-11-19:
Don't see how firmware compression would cause this

laris, 2023-11-20:
Any troubleshooting and log can track such issue?

laris, 2023-12-08:
I don't know how to confirm the root cause even Robert help me to build the EFI from scratch.
I think I can bear the blackscreen and seconds to wait to login page come out.
let's close this issue.

closed 2023-12-08 by laris

VisualEhrmanntraut, 2023-12-08:
Have you tried using Release build instead of Debug? @laris

laris, 2023-12-08:
> Have you tried using Release build instead of Debug? @laris

You are right.
I download the release kext and found no issue now.
Should my fault to use debug version
Niubility!

`NootedRed-ci1062-20231203-f39b450-RELEASE.kext`