ChefKissInc/NootedRed #194: Bootloop on macOS 14 Sonoma

frost-1256, 2023-11-02:
**Describe the bug**
On macOS 14 Sonoma,the login screen is a bit choppy, etc., and the system reboots a few seconds after logging in.
With Ventura and Sonoma Beta, there was no problem.

**To Reproduce**
Steps to reproduce the behavior:
Booting with NootedRed enabled in Sonoma

**Expected behavior**

**Screenshots**

**System Information:**
 - macOS Version: macOS 14 Sonoma 14.1 23B74
 - CPU: Ryzen 3 3200G

**Additional context**
4GB of memory was configured from UEFI.

frost-1256, 2023-11-02:
I am currently testing again and it reboots with the Apple logo after the verbose log!

frost-1256, 2023-11-02:
When I waited a bit and logged in, a panic repair boot
message appeared and rebooted after a few seconds
[image]

pmnlla, 2023-11-06:
Managed to recreate. Will upload logs later. 
For future reference, what version of NRed are you running?

ethernetme, 2023-11-11:
My kexts are updated, but I'm still getting this issue

VisualEhrmanntraut, 2023-11-11:
send your config with the serials stripped

VisualEhrmanntraut, 2023-11-11:
> the login screen is a bit choppy

Probably TSC sync issue

ethernetme, 2023-11-11:
> send your config with the serials stripped

Here it is. If you find a fix, I wouldn't be able to test it though, since I'm already reinstalling Ventura. I didn't get the issue on that OS, only after updating to Sonoma.

[config.zip](https://github.com/ChefKissInc/NootedRed/files/13325823/config.zip)

VisualEhrmanntraut, 2023-11-11:
> Probably TSC sync issue

Ditto.

VisualEhrmanntraut, 2023-11-11:
Closing due to unwillingness to resolve issue.

closed 2023-11-11 by VisualEhrmanntraut

pmnlla, 2023-11-11:
So what should i do to solve the issue?

VisualEhrmanntraut, 2023-11-11:
Configure your system properly.

ethernetme, 2023-11-11:
what do you think we've doing hun

VisualEhrmanntraut, 2023-11-11:
Not that. :-) I suggest you retry with the commit I pushed a bit ago.

pmnlla, 2023-11-11:
> Configure your system properly.

Yes I'm well aware there's a potential issue with our configs. I looked up TSC sync in Dortania's guides and only found an ssdt that works for Intel cpus. What do you want me to try?

pmnlla, 2023-11-13:
Seeking an update on this. AmdTscSync did not fix the issue.

pmnlla, 2023-11-30:
Seeking an update on this again. Tried adding AmdTscSync to my kexts, that did not fix it.

ethernetme, 2023-12-28:
> Not that. :-) I suggest you retry with the commit I pushed a bit ago.

Weeks later, I'm still getting this issue in Sonoma even with a clean install and the latest version of the kext. The installation went through just fine but it keeps bootlooping on the first boot screen. Would you mind helping me find the issue? Here is my config file if needed:
[config.zip](https://github.com/ChefKissInc/NootedRed/files/13787842/config.zip)