ChefKissInc/NootedRed #197: Just stop on loading screen

DepsCian, 2023-11-10:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Boot with latest build NRed
2. Profit

**Expected behavior**
Desktop

**Screenshots**
[image]

**System Information:**
 - macOS Version: MacOS 13.6
 - CPU: Ryzen 3 2200G

**Additional context**
On older builds I do not have this problem, but there is a problem with black screen.

VisualEhrmanntraut, 2023-11-11:
@DepsCian show your boot args

DepsCian, 2023-11-12:
@VisualEhrmanntraut npci=0x3000 alcid=11 maxmem=13000

VisualEhrmanntraut, 2023-11-19:
@DepsCian have you tried the latest commit?

nanohelperDev, 2023-11-23:
just back to windows 11 its better than wasting time trying to use something not usable on your machine , you just wasting time

gorouflex, 2023-11-23:
> just back to windows 11 its better than wasting time trying to use something not usable on your machine , you just wasting time

[image]

VisualEhrmanntraut, 2023-12-02:
ERR_CONNECTION_TIMED_OUT

DepsCian, 2023-12-23:
After some time when I was on vacation I still tried the latest build but still sometimes I only see a black screen. Note: I use HDMI and Sonoma