ChefKissInc/NootedRed #199: Erratic colours on 5700G + Big Sur

rtd1250, 2023-11-17:
**Describe the bug**
Colour is not correct on HDMI output with NootedRed enabled.

**To Reproduce**
Steps to reproduce the behavior:
1. Enable NootedRed
2. Reboot
3. Be greeted with LSD desktop

**Expected behavior**
Colours should be displayed correctly, as in black should be black instead of green.

**Screenshots**
[image]

**System Information:**
 - macOS Version: macOS 11.7.10
 - CPU: AMD Ryzen 5 5700G
 - Motherboard: MSI X570
 - SMBIOS: iMacPro1,1

**Additional context**
Changing resolutions and colour depth doesn't really seem to do anything. This only happens on HDMI output, screenshots look fine (the screenshot above was taken with an HDMI grabber). The acceleration itself does work and in the About window it reports as "AMD Radeon Graphics 1 GB".

VisualEhrmanntraut, 2023-11-19:
Duplicate of #58

VisualEhrmanntraut, 2023-11-19:
Oops, wrong number

VisualEhrmanntraut, 2023-11-19:
Duplicate of #65

closed 2023-11-19 by VisualEhrmanntraut