ChefKissInc/NootedRed #200: Display blur and crash with terminal application  "alacritty"

laris, 2023-11-19:
When open the alacritty, screen blur and system hung and crash.

https://github.com/alacritty/alacritty
- Require: At least OpenGL ES 2.0
--------------------------------------------------------------------------------

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**System Information:**
 - macOS Version: [e.g. macOS 11.7.4]
 - CPU: [e.g. AMD Ryzen 5 3600G]

**Additional context**
Add any other context about the problem here.

VisualEhrmanntraut, 2023-11-19:
Duplicate of #158

VisualEhrmanntraut, 2023-11-19:
Duplicate of #65

closed 2023-11-19 by VisualEhrmanntraut

laris, 2023-11-19:
Need I upload the diagnosticlog here for analysis?

VisualEhrmanntraut, 2023-11-19:
Sure! Crash logs would be nice as well :)

laris, 2023-11-19:
[2023.11.19-diagnosticreports.zip](https://github.com/ChefKissInc/NootedRed/files/13403142/2023.11.19-diagnosticreports.zip)

I forgot to upload and found you marked it closed :)

Thanks for your works!

ExtremeXT, 2023-11-19:
@laris It looks like you're on an old version of NootedRed.

laris, 2023-11-20:
I checked opencore config, it should be ci1055 20231112 dfd2ffa commit.
let me try 20231117-957505c-ci1057

laris, 2023-11-20:
update to latest and now will test again.

```

/Volumes/EFI/EFI/OC/Kexts/NootedRed-ci1057-20231117-957505c-DEBUG.kext

kextstat -b com.ChefKissInc.NootedRed
Executing: /usr/bin/kmutil showloaded --bundle-identifier com.ChefKissInc.NootedRed
No variant specified, falling back to release
Index Refs Address            Size       Wired      Name (Version) UUID <Linked Against>
   59    0 0xffffff8003c9f000 0x114000   0x114000   com.ChefKissInc.NootedRed (1.0.0) 2405C963-EC91-3C81-9A1D-8FB2F68556FC <58 18 9 7 6 3 2 1>
```

laris, 2023-11-20:
> @laris It looks like you're on an old version of NootedRed.

Could you share how to double confirm the driver version? I confirm according opencore config ref https://github.com/ChefKissInc/NootedRed/issues/200#issuecomment-1818104578

I can provide more detail or log if you need.

laris, 2023-11-20:
https://github.com/ChefKissInc/NootedRed/assets/951666/5ff1b4dd-9c79-46bb-aae8-93a56c64e938

upload video, after open the alacritty, the dock status.

VisualEhrmanntraut, 2023-11-20:
Same problem as Sublime Text. The only thing you can do is wait for a fix and/or disable HW acceleration for the app temporarily

laris, 2023-11-24:
thank you.
```
./alacritty -o debug.renderer=gles2_pure -v
Created log file at "/var/folders/py/_4bld1296cx59mt3l3wqnv7h0000gn/T/Alacritty-1354.log"
[0.000009558s] [INFO ] [alacritty] Welcome to Alacritty
[0.000272171s] [INFO ] [alacritty] Version 0.12.3 (5efb069)
[0.000670589s] [INFO ] [alacritty_config_derive] Config import not found:
[0.001092822s] [INFO ] [alacritty] Configuration files loaded from:
2023-11-24 08:44:30.111 alacritty[1354:20366] WARNING: Secure coding is not enabled for restorable state! Enable secure coding by implementing NSApplicationDelegate.applicationSupportsSecureRestorableState: and returning YES.
[0.031249001s] [INFO ] [alacritty] Using Apple CGL
[0.078801984s] [INFO ] [alacritty] Window scale factor: 2
[0.106355648s] [INFO ] [alacritty] Running on AMD Radeon Graphics OpenGL Engine
[0.106441899s] [INFO ] [alacritty] Using OpenGL ES 2.0 renderer
[0.117393902s] [INFO ] [alacritty] Cell size: 19 x 44
[0.117465807s] [INFO ] [alacritty] Padding: 0 x 0
[0.117493580s] [INFO ] [alacritty] Width: 2280, Height: 2200
[0.119880180s] [INFO ] [alacritty] PTY dimensions: 50 x 120
[0.123276487s] [INFO ] [alacritty] Initialisation complete
[0.155839656s] [INFO ] [alacritty] Padding: 0 x 0
[0.155955574s] [INFO ] [alacritty] Width: 2280, Height: 1592
```

VisualEhrmanntraut, 2023-11-24:
that's a macOS OpenGL driver bug with the texture display. even mesa has the same issue on OpenGL 2 renderer