ChefKissInc/NootedRed #203: All application windows cannot be rendered properly after increased iGPU vRAM to 1g/2g

itzdrli, 2023-11-26:
### macOS Version

Sonoma

### What is your CPU's model?

AMD R7 5800H

### Please describe the behaviour in detail.

after using the nootedred with only igpu, i was facing performance issue: screen freeze for seconds or minutes, laggy actions etc. Someone told me try to increase vRAM via Somkeless-UMAF, this is what I did: 
[image]
and this is what happend after I logged in and opened some window, my screen truned to be like this:
[image]

### What should've happened instead?

All application windows should be rendered properly

### Bug report submission terms

- [X] I am using the latest version of NootedRed.
- [X] I am on the latest minor and patch version of macOS (ex. 7 is the minor and 10 is the patch of macOS 11.7.10).
- [X] The bug only happens with NootedRed enabled.
- [X] I have ruled out any external factors that could interfere, including, but not limited to, using OCLP or WhateverGreen.

VisualEhrmanntraut, 2023-11-26:
***Which app?***

itzdrli, 2023-11-26:
> _**Which app?**_

all apps, including the top bar of the system

VisualEhrmanntraut, 2023-11-26:
Which app did you launch before that happened

itzdrli, 2023-11-26:
> Which app did you launch before that happened

google chrome, 1password, about this mac, settings

VisualEhrmanntraut, 2023-11-26:
Google Chrome and 1Password are both OpenGL apps.

itzdrli, 2023-11-26:
> Google Chrome and 1Password are both OpenGL apps.

but other apps are acting same, even I closed chrome and 1pass

VisualEhrmanntraut, 2023-11-26:
Because the VRAM got corrupted after you launched them

itzdrli, 2023-11-26:
> Because the VRAM got corrupted after you launched them

so is there any fix for this situation?

VisualEhrmanntraut, 2023-11-26:
Just restart your machine and delete the apps for now

closed 2023-11-26 by VisualEhrmanntraut