ChefKissInc/NootedRed #205: Laptop freezes and restarts quickly within 10 seconds after booting

aloums, 2023-12-02:
### macOS Version

Sonoma

### What is your CPU's model?

Ryzen 3500U

### Please describe the behaviour in detail.

I've never encountered this on a previous system

### What should've happened instead?

Start the system normally

### Bug report submission terms

- [X] I am using the latest version of NootedRed.
- [X] I am on the latest minor and patch version of macOS (ex. 7 is the minor and 10 is the patch of macOS 11.7.10).
- [X] The bug only happens with NootedRed enabled.
- [X] I have ruled out any external factors that could interfere, including, but not limited to, using OCLP or WhateverGreen.

SiddharthManthan, 2023-12-04:
Try disabling changing (dynamic ?) wallpaper. Someone else had this issue too and suggested me to do the same.

VisualEhrmanntraut, 2023-12-06:
Sounds like the TSC is unsynced, assuming the first submission term agreed upon is true. This kext will likely help: https://github.com/Seey6/CpuTscSync

aloums, 2023-12-10:
> 听起来 TSC 未同步，假设商定的第一个提交条款是真实的。这个 kext 可能会有所帮助：[https ://github.com/Seey6/CpuTscSync](https://github.com/Seey6/CpuTscSync)

VisualEhrmanntraut, 2023-12-10:
?

VisualEhrmanntraut, 2023-12-21:
Timed out waiting for response.

closed 2023-12-21 by VisualEhrmanntraut

pmnlla, 2023-12-27:
> Try disabling changing (dynamic ?) wallpaper. Someone else had this issue too and suggested me to do the same.

This also worked for me, but doesn't solve the root cause of the issue.

ho1yspirt, 2024-01-20:
Getting it, when starting the browser. In case of Firefox, the laptop reboots when starting it, in case of Chrome when tapping the address bar, some red color artifacts are popping up.