ChefKissInc/NootedRed #206: Internal display not working only HDMI output

pego149, 2023-12-04:
### macOS Version

Sonoma

### What is your CPU's model?

Ryzen 9 4900HS

### Please describe the behaviour in detail.

When I want to boot on my ASUS zephyrus G15 GA502IV with enabled Nootedred.kext I am not able find my internal screen. Only HDMI output is successfully recognized (doesn't matter if it's connected) internal display on laptop is always black (turned off). When I disabled nootedred the screen works but as unknown display. HW acceleration works well with Nootedred.kext. Thank you.

### What should've happened instead?

Internal display should work. I don't care about HDMI.

### Bug report submission terms

- [X] I am using the latest version of NootedRed.
- [X] I am on the latest minor and patch version of macOS (ex. 7 is the minor and 10 is the patch of macOS 11.7.10).
- [X] The bug only happens with NootedRed enabled.
- [X] I have ruled out any external factors that could interfere, including, but not limited to, using OCLP or WhateverGreen.

cornshiba, 2023-12-04:
are you using the `-NRedDPDelay` boot-arg? If no, can you please try adding it and then try again to see if you get video output on your internal display?

pego149, 2023-12-04:
Yes I tried it with this boot-arg option but nothing changed.

pego149, 2023-12-05:
I am attaching agdcdiagnose log when HDMI is disconnected, so I can see in connections there is "DCC Failed". Maybe this can be issue.
[agdcdiagnose.txt](https://github.com/ChefKissInc/NootedRed/files/13557727/agdcdiagnose.txt)

VisualEhrmanntraut, 2023-12-06:
> I am using the latest version of NootedRed.

Is this actually true or did you just click it to make the issue

VisualEhrmanntraut, 2023-12-06:
Misclick

pego149, 2023-12-07:
Hi, yes I tried many builds of nootedred next, but yes I am sure that I tried last, too.

pego149, 2023-12-07:
Hi @jalavoui I tried build from your post but without any success. I am attaching journal file from booting with drm debug boot flag from my kali linux: 
[journal.txt](https://github.com/ChefKissInc/NootedRed/files/13606351/journal.txt)

VisualEhrmanntraut, 2024-01-02:
@pego149 is your BIOS up to date

metronormalist, 2024-01-15:
Hi, I also have a G14 Zephyrus and I follow the guide from this guy's [repo](https://github.com/PIut02/ROG-Zephyrus-G14-GA401-Hackintosh). You could check the EFI out there and rebuild your hackintosh. I could confirm it works fine for me with the 2020 G14. 
For me, the `Ventura` works better than the `Sonoma`, so if you want to use it, remember to switch to the `Ventura` branch as the `main` branch is for `Sonoma`. In my case, it's the 2020 model so I get the EFI inside the `2020` folder, I don't which model you have so you could give it a shot with both the `2020` and `2021`.

pego149, 2024-01-18:
> @pego149 is your BIOS up to date

Hi, yes my BIOS is up to date

> Hi, I also have a G14 Zephyrus and I follow the guide from this guy's [repo](https://github.com/PIut02/ROG-Zephyrus-G14-GA401-Hackintosh). You could check the EFI out there and rebuild your hackintosh. I could confirm it works fine for me with the 2020 G14. For me, the `Ventura` works better than the `Sonoma`, so if you want to use it, remember to switch to the `Ventura` branch as the `main` branch is for `Sonoma`. In my case, it's the 2020 model so I get the EFI inside the `2020` folder, I don't know which model you have so you could give it a shot with both the `2020` and `2021`.

Hi, thank you for answering. Yes I tried this repo as source of my EFI folder but still without success. If I disable nootedred.kext the internal display works normally but without acceleration. When I enable nootedred.kext the internal display powers off during boot and then the HDMI display turns on and everything is shown only there (doesn't matter if hdmi cable is connected or not)

metronormalist, 2024-01-18:
> Hi, yes my BIOS is up to date
>
>
>
>
>
>
>
>
> Hi, thank you for answering. Yes I tried this repo as source of my EFI folder but still without success. If I disable nootedred.kext the internal display works [...]

That's weird to hear. Maybe your model is not supported well? I think you should wait for updates from `NootedRed` as they will be released soon to fix remaining issues and may support some systems as yours.

VisualEhrmanntraut, 2024-03-25:
Please retry with the latest commits

pego149, 2024-03-29:
Hi @VisualEhrmanntraut thank you very much. Last commit works well and finally my laptop booted with working screen.

closed 2024-03-29 by VisualEhrmanntraut