ChefKissInc/NootedRed #209: Opening Balena Etcher hangs the whole system for 1 minute

Mirarchi, 2023-12-10:
### macOS Version

Sonoma

### What is your CPU's model?

AMD Ryzen 7 4800H

### Please describe the behaviour in detail.

When opening galena etcher, the system hangs for a minute, then it seems to behave normally.
Using the latest kext available under Actions.
Don't know if it could be related to #13
 
Anyway thanks for the huge work.
Best regards

### What should've happened instead?

Opening balena etcher should be a matter of few seconds

### If applicable, attach the .gpuRestart, .panic, etc file related to this issue.

[Kernel_2023-12-10-113655_MacBook-Pro.gpuRestart.txt](https://github.com/ChefKissInc/NootedRed/files/13627428/Kernel_2023-12-10-113655_MacBook-Pro.gpuRestart.txt)
[Kernel_2023-12-10-113705_MacBook-Pro.gpuRestart.txt](https://github.com/ChefKissInc/NootedRed/files/13627430/Kernel_2023-12-10-113705_MacBook-Pro.gpuRestart.txt)
[Kernel_2023-12-10-113716_MacBook-Pro.gpuRestart.txt](https://github.com/ChefKissInc/NootedRed/files/13627431/Kernel_2023-12-10-113716_MacBook-Pro.gpuRestart.txt)
[Kernel_2023-12-10-113717_MacBook-Pro.gpuRestart.txt](https://github.com/ChefKissInc/NootedRed/files/13627432/Kernel_2023-12-10-113717_MacBook-Pro.gpuRestart.txt)
[Kernel_2023-12-10-113727_MacBook-Pro.gpuRestart.txt](https://github.com/ChefKissInc/NootedRed/files/13627433/Kernel_2023-12-10-113727_MacBook-Pro.gpuRestart.txt)

### Bug report submission terms

- [X] I am using the latest version of NootedRed.
- [X] I am on the latest minor and patch version of macOS (ex. 7 is the minor and 10 is the patch of macOS 11.7.10).
- [X] The bug only happens with NootedRed enabled.
- [X] I have ruled out any external factors that could interfere, including, but not limited to, using OCLP or WhateverGreen.

VisualEhrmanntraut, 2023-12-10:
Duplicate of #158

closed 2023-12-10 by VisualEhrmanntraut

Mirarchi, 2023-12-10:
Thanks for immediately pointing back to the duplicate, I see a lot of work in progress so maybe I will wait for an 'official' update of the kext.

Thanks