ChefKissInc/NootedRed #213: Black screen after a long sleep, but the system works

lev666, 2023-12-24:
### macOS Version

Sonoma

### What is your CPU's model?

AMD Ryzen 7 5800H

### Please describe the behaviour in detail.

Good afternoon. Noticed this problem quite some time ago, what is the problem? It lies in the fact that if the laptop is in sleep mode for more than an hour, then when you wake up the screen will be black, but the system will work (I checked this by the fact that it emitted various sounds when pressing various keys on the keyboard and simultaneously with this screen was black).

Provided was a panic log .gpuRestart but it never appeared after a long sleep of the laptop, but the screen never worked after a long sleep.

### What should've happened instead?

Other than ruling out other components (that they are the culprit), I haven't done anything else.

### Bug report submission terms

- [X] I am using the latest version of NootedRed.
- [X] I am on the latest minor and patch version of macOS (ex. 7 is the minor and 10 is the patch of macOS 11.7.10).
- [X] The bug only happens with NootedRed enabled.
- [X] I have ruled out any external factors that could interfere, including, but not limited to, using OCLP or WhateverGreen.

lev666, 2023-12-25:
Updating AMDTscSync to Seey6/CpuTscSync did not help either. The system wakes up after a long sleep, responds to keys, but the screen is black.

lev666, 2023-12-26:
> > Updating AMDTscSync to Seey6/CpuTscSync did not help either. The system wakes up after a long sleep, responds to keys, but the screen is black.
> 
> Try adding the 'Disable RTC wake scheduling kext patch'; it will disable darkwake，may solve the issue

Good evening. Thank you for your suggestion, but it is also not a solution.

typeriter, 2024-01-08:
I can confirm the same exact issue occurring for me with Ryzen 7 5825 running Ventura. It only seems to happen after a long sleep. I can hear sounds and confirmed that the machine is alive because I had TextEdit open, and saw the gibberish I typed trying to wake it up after I held down the power button and rebooted. Pressing ESC, which usually turns the display on and off from the lock screen, does nothing when this error occurs. For me, this is happening on a Lenovo Thinkpad E14 G4 laptop. I appreciate the work being done on NootedRed. Thank you.

aidanbae123, 2024-01-09:
I can also confirm the same issue with a ThinkPad E14 G4 running a Ryzen 7 5825U.

metronormalist, 2024-01-15:
I  my case, the lockscreen appears normally but the keyboard does not work although the trackpad/mouse works. I've tried this method and I could confirm this works for me. You guys could give it a shot:
- Use this tool - [UMAF](https://github.com/DavidS95/Smokeless_UMAF) to configure some of your BIOS settings. Just put the files into your USB flash drive after you format it and just boot into it.
- Select `Device Manager` when the menu shows up.
- Select `AMD PBS` and a lot of options show up, just move down until you see the `S3/Modern Standby Wake Up` option, change it from `Off` or `S3` to `Modern Standby Wake Up`.
- Press `ESC` to return, remember to press `Y` to save your changes.
You guys can test it out, in my case, it works perfectly.

lev666, 2024-01-15:
> I  my case, the lockscreen appears normally but the keyboard does not work although the trackpad/mouse works. I've tried this method and I could confirm this works for me. You guys could give it a [...]

Good afternoon, thanks for this suggestion, but alas, with me it only results in the system not even trying to put the laptop to sleep:(

metronormalist, 2024-01-15:
> Good afternoon, thanks for this suggestion, but alas, with me it only results in the system not even trying to put the laptop to sleep:(

Yeah, I got you. So maybe you need to configure the DSDT or kexts to make it sleep. I don't know much about that because I use an existing EFI from a guy so you need to search to learn more about your settings. I recommend you follow the [Dortania](https://dortania.github.io/OpenCore-Install-Guide/) guide. In addition, maybe you could try to run these commands in the `Terminal` to see if it works for you:
```
sudo pmset autopoweroff 0
sudo pmset powernap 0
sudo pmset standby 0
sudo pmset proximitywake 0
sudo pmset tcpkeepalive 0
```

amane1234, 2024-01-30:
I used to have a same problem with using NAVI discrete graphics with opencore.

This problem was resolved by adding "whatevergreen.kext" in kext folder and "boot-arg : agdpmod=pikera" in config.plist

After that, no more blackscreen after waking up from S3 and S4

Since I was able to resolve this problem with whatevergreen.kext with boot-arg : agdpmod=pikera method,

I recommand to follow aforementioned steps first.

If using whatevergreen.kext with nootedred.kext introduce some unstabilties, crashes, or glitches,

you may replace "board-id” into “board-ix" manually (same method with whatevergreen.kext) by following these steps: https://pikeralpha.wordpress.com/2015/11/23/patching-applegraphicsdevicepolicy-kext/

I hope your problem will be fixed soon!

VisualEhrmanntraut, 2024-01-30:
[image]

Mazi29, 2024-02-23:
> I can confirm the same exact issue occurring for me with Ryzen 7 5825U running Ventura. It only seems to happen after a long sleep. I can hear sounds and confirmed that the machine is alive because I had TextEdit open, and saw the gibberish I typed trying to wake it up after I held down the power button and rebooted. Pressing ESC, which usually turns the display on and off from the lock screen, does nothing when this error occurs. For me, this is happening on a Lenovo Thinkpad E14 G4 laptop. I appreciate the work being done on NootedRed. Thank you.

I did a test on my system, and noticed that the issue happens when my system wakes from sleep, but the display doesn’t wake as the lid is closed and then it goes back to sleep. Any attempt to wake up the display totally fails. You can recreate this by putting the computer to sleep using the Apple logo menu, (as my system requires 2 key presses to, one to wake the system up, and another to wake the display up, ) I press the key to wake the system without waking the display

The article that solves this problem is located here, it disable scheduled pm wakes and it shows the optimal settings for power etc

https://www.tonymacx86.com/threads/solved-ventura-sonoma-random-scheduled-pm-wake-from-sleep.323359/

lev666, 2024-02-25:
> I did a test on my system, and noticed that the issue happens when my system wakes from sleep, but the display doesn’t wake as the lid is closed and then it goes back to sleep. Any attempt to wake [...]

Good afternoon, thank you for your suggested post on the link. The tips in this post only partially helped with the long sleep fix, namely, the system now when sleeping for a very long time, after waking it up, does not go into panic and then reboot. Now the system continues to function, but the black screen problem still persists.

aidanbae123, 2024-02-25:
> I did a test on my system, and noticed that the issue happens when my system wakes from sleep, but the display doesn’t wake as the lid is closed and then it goes back to sleep. Any attempt to wake [...]

This actually made my problem much worse. Previously, my system would wake if the sleep duration was under around 2 hours, but now it doesn't wake up at all.