ChefKissInc/NootedRed #214: Bootloop on Sonoma

ethernetme, 2023-12-28:
### macOS Version

Sonoma

### What is your CPU's model?

Ryzen 5 3500U

### Please describe the behaviour in detail.

Sonoma crashes and reboots a few seconds after booting. Even with a clean install, the installation goes well but it starts crashing on the first boot screen. This issue only happens on Sonoma specifically and with the NootedRed kext enabled.

### What should've happened instead?

The system should work fine without any crashes.

### Bug report submission terms

- [X] I am using the latest version of NootedRed.
- [X] I am on the latest minor and patch version of macOS (ex. 7 is the minor and 10 is the patch of macOS 11.7.10).
- [X] The bug only happens with NootedRed enabled.
- [X] I have ruled out any external factors that could interfere, including, but not limited to, using OCLP or WhateverGreen.

ethernetme, 2023-12-28:
> enable all AMD Vanilla kernel patch

I've enabled all the patches but that didn't fix the issue. Still, I don't see how that'd fix anything since everything works fine without the NootedRed kext.

ethernetme, 2023-12-28:
> because i have a same problem since ventura 13.4+,sonoma. I enable all kernel patch to fix it.

Unfortunately, it's not the case for me. If you know any other way to fix it, please let me know.

c13MrT, 2023-12-30:
I think it's related to your specific setup. It works for me in the latest Sonoma version. Without getting more input from you, it doesn't sound like it's a bug.

ethernetme, 2023-12-30:
> I think it's related to your specific setup. It works for me in the latest Sonoma version. Without getting more input from you, it doesn't sound like it's a bug.

I don't think that's the case since this only happens in Sonoma, and I'm not the only person with this issue. Besides, I don't see what could be wrong with my setup. I'll attach my config file here just in case it's helpful.
[config.zip](https://github.com/ChefKissInc/NootedRed/files/13797910/config.zip)

VisualEhrmanntraut, 2024-01-02:
Cannot reproduce. Join the Telegram for support.

closed 2024-01-02 by VisualEhrmanntraut

ethernetme, 2024-01-03:
> Cannot reproduce. Join the Telegram for support.

Yes I'm already there but I don't know how to get support without getting ignored

VisualEhrmanntraut, 2024-01-03:
You don't appear to actually be in our Telegram.