ChefKissInc/NootedRed #217: boot loop when nootedred.text enabled

CrotchBurnt, 2024-01-10:
### macOS Version

Sonoma

### What is your CPU's model?

AMD Ryzen 7 5825U

### Please describe the behaviour in detail.

It works well while installing but it turns into forever boot loop even if only Lilu、VirtualSMC、Nootedred enabled. When I disable NRed only, I could enter the system. 
[image]

### What should've happened instead?

Booting into the system

### Bug report submission terms

- [X] I am using the latest version of NootedRed.
- [X] I am on the latest minor and patch version of macOS (ex. 7 is the minor and 10 is the patch of macOS 11.7.10).
- [X] The bug only happens with NootedRed enabled.
- [X] I have ruled out any external factors that could interfere, including, but not limited to, using OCLP or WhateverGreen.

VisualEhrmanntraut, 2024-01-10:
https://chefkissinc.github.io/nred#prerequisites

CrotchBurnt, 2024-01-10:
> https://chefkissinc.github.io/nred#prerequisites

Yes, I checked all that, with VRAM of 512MB. I just updated panic file

CrotchBurnt, 2024-01-12:
probably plist has compromised by OCAT, I started from the start all over again with ProperTree, everything's fine.

closed 2024-02-02 by VisualEhrmanntraut