ChefKissInc/NootedRed #219: Weird cursor behavior

Wangxxis, 2024-01-24:
### macOS Version

Ventura

### What is your CPU's model?

R7 5800H

### Please describe the behaviour in detail.

This paragraph is translated from Chinese, when I switch the pointer from the finger state to the pointer, there will be a flicker situation, is this normal? I found in a Google search others have similar problems, https://forums.macrumors.com/threads/weird-cursor-behavior.2230749/
https://apple.stackexchange.com/questions/393951/jittery-macos-mouse-cursor-when-hovering-over-links

这段话是通过中文翻译而来，当我指针从手指状态切换成指针时，会出现闪烁情况，这是正常的吗？我通过google搜索发现其他人也有类似的问题。

### What should've happened instead?

No flicker

### Bug report submission terms

- [X] I am using the latest version of NootedRed.
- [X] I am on the latest minor and patch version of macOS (ex. 7 is the minor and 10 is the patch of macOS 11.7.10).
- [X] The bug only happens with NootedRed enabled.
- [X] I have ruled out any external factors that could interfere, including, but not limited to, using OCLP or WhateverGreen.

VisualEhrmanntraut, 2024-01-24:
Am I supposed to fix Apple's OS?

closed 2024-01-24 by VisualEhrmanntraut

Wangxxis, 2024-01-25:
> Am I supposed to fix Apple's OS?
Sorry to bother you,Turns out this is a Apple's OS problem