ChefKissInc/NootedRed #221: All DAWs crash when opening some audio plugins. i.e: Waves CLA MixHub

leociotti, 2024-02-01:
### macOS Version

Catalina

### What is your CPU's model?

AMD Ryzen 7 5800U

### Please describe the behaviour in detail.

I've tried several DAWs (Digital Audio Workstations) like Pro Tools, Cubase, Reaper, Studio One, Logic Pro X, and all of them crashed and closed when I tried opening up some Waves plugins, like the CLA MixHub or CLA Bass. They just close, without any

### What should've happened instead?

The plugins should open.
Attached is the zipped ips file. (couldn't attach the file without zipping it.)
[Studio One-2024-02-01-163738.ips.zip](https://github.com/ChefKissInc/NootedRed/files/14129995/Studio.One-2024-02-01-163738.ips.zip)

### If applicable, attach the .gpuRestart, .panic, etc file related to this issue.

[Crash Report.pdf](https://github.com/ChefKissInc/NootedRed/files/14130017/Crash.Report.pdf)

### Bug report submission terms

- [X] I am using the latest version of NootedRed.
- [X] I am on the latest minor and patch version of macOS (ex. 7 is the minor and 10 is the patch of macOS 11.7.10).
- [X] The bug only happens with NootedRed enabled.
- [X] I have ruled out any external factors that could interfere, including, but not limited to, using OCLP or WhateverGreen.

VisualEhrmanntraut, 2024-02-01:
Intel MKL incompatibility with AMD, not a NootedRed issue.
[image]

leociotti, 2024-02-01:
Sorry, I really didn't know that. Thank you for clarifying it to me.

closed 2024-02-02 by VisualEhrmanntraut