ChefKissInc/NootedRed #224: NootedRed black screen ASUS A320M-K + AMD Athlon 3000G int Radeon Vega 3

dipol0, 2024-02-03:
### macOS Version

Ventura

### What is your CPU's model?

AMD Athlon 3000G

### Please describe the behaviour in detail.

Mobo - ASUS PRIME A320M-K
CPU — AMD Athlon 3000G with Radeon Vega 3
Display connected over D-SUB (VGA).

May be important? -> NVME storage inserted to PCI-E slot over adapter (my cpu don't have pci-e lines on original m2 connector)

Bootloader - OpenCore
Using NootedRed.kext v1.0.0-RELEASE

BIOS:

    Above 4Gb - Enabled
    CSM - Disabled
    IOMMU - Disabled
    Serial Port - Disbled
    VT-D - Enabled
    Multimonitor - Disabled
    Primary Video - IGFX
    OC boot-args: npci=0x2000 unfairgva=1 alcid=11 -v debug=0x100 keepsyms=1

After boot strings display blank (black screen). But over ~10..20 reboot/shutdowns it may "fly up" and work. I try more things… change in BIOS Multi monitor option. Try connect TV over HDMI...playing with boot-args… but don’t have good result. Very randomly OS booting normal and work, but more frequently just black screen.

But...then black screen PC with MacOS can be ping and i can login over SSH. Access over VNC (shared display) don't work.

How solve this trouble?

*******************************
Proof sometime (randomly...) it work:
[image]

### What should've happened instead?

I can see logon screen.

### Bug report submission terms

- [X] I am using the latest version of NootedRed.
- [X] I am on the latest minor and patch version of macOS (ex. 7 is the minor and 10 is the patch of macOS 11.7.10).
- [X] The bug only happens with NootedRed enabled.
- [X] I have ruled out any external factors that could interfere, including, but not limited to, using OCLP or WhateverGreen.

VisualEhrmanntraut, 2024-02-04:
`-NRedDPDelay` is intended for internal eDP screens which suffer from a link training problem. macOS does not support VGA, so I assume your motherboard is internally converting the signal to something like DP or DVI. Assuming this, then macOS is having trouble with this internal adaptor.

dipol0, 2024-02-04:
i see most frequently i have normal boot if just fully unplug power cord. Wait a 30..60 sec. Insert it back and start PC. But 1..2 from 5 starts may be fails with blackscreen. others normal boot. And... my old monitor have only D-SUB(VGZ) and DVI IN's. I connect it to mobo over hdmi cale use convrter hdmi-dvi.

Continue watching... ))

dipol0, 2024-02-05:
> you're lucky to use the pro ASUS A320M (like myself) try the version i posted on insanelymac. i'm on ventura don't ask me for sonoma

Same use Venrura. See PM on insanelymac. Tnx