ChefKissInc/NootedRed #225: HDMI Sound Appear but No Sound

yeremia1234, 2024-02-04:
### macOS Version

Ventura

### What is your CPU's model?

AMD Ryzen 5 3500u

### Please describe the behaviour in detail.

I've already install nootedred kext on my mac. The graphic is working, but i have a problem. HDMI Audio appear on setting but no sound when connected to TV. Any solution?

### What should've happened instead?

Fix that i trust in you

### Bug report submission terms

- [X] I am using the latest version of NootedRed.
- [X] I am on the latest minor and patch version of macOS (ex. 7 is the minor and 10 is the patch of macOS 11.7.10).
- [X] The bug only happens with NootedRed enabled.
- [X] I have ruled out any external factors that could interfere, including, but not limited to, using OCLP or WhateverGreen.

dipol0, 2024-02-04:
Hi. Very strange. I have more simple CPU - AMD Athlon 3000G with int Radeon Vega 3 and mobo ASUS PRIME A320M-K.
When i boot MacOS 13 Ventura all audio out works - and line out on backside panel and if i connect old TV Samsung over hdmi cable. Check settings sound - may be on your system Line Out as default?

yeremia1234, 2024-02-04:
> Hi. Very strange. I have more simple CPU - AMD Athlon 3000G with int Radeon Vega 3 and mobo ASUS PRIME A320M-K. When i boot MacOS 13 Ventura all audio out works - and line out on backside panel and if i connect old TV Samsung over hdmi cable. Check settings sound - may be on your system Line Out as default? When i think this kexst project orient on graphics, no sound.

I set audio default on samsung but the sound is not appear. Only graphic show on tv but sound not show

dipol0, 2024-02-04:
> I set audio default on samsung but the sound is not appear. Only graphic show on tv but sound not show

You TV correct shown in Displays? You see HDMI as out on you TV in Sound?
[image]

And! )) may be you just set low level volume on TV or mute it ? )) Check. ;)

+ in my config.plist module for audio set as 4 + in boot-args alcid=11:
[image]
[image]

VisualEhrmanntraut, 2024-02-04:
It is actually a known issue and it has no known cause or solution. The problem only appears on a very small amount of devices.

yeremia1234, 2024-02-04:
![Screenshot 2024-02-04 at 19 34 19](https://github.com/ChefKissInc/NootedRed/assets/31928254/736c6
[image]
f36-2f12-40df-a533-d00978bc3b66)

this is the screenshot but i dont hear any sound on tv. Only picture

yeremia1234, 2024-02-04:
[image]

VisualEhrmanntraut, 2024-02-04:
> It is actually a known issue and it has no known cause or solution. The problem only appears on a very small amount of devices.

.

dipol0, 2024-02-04:
@yeremia1234 check cable on another devices - its work audio ? If you TV have few hdmi in - try another and in boot-arg try `aclid=1`

VisualEhrmanntraut, 2024-02-04:
Ahem. Let's see if you can ignore the developer (me) now.

> It is actually a known issue and it has no known cause or solution. The problem only appears on a very small amount of devices.

We have already tried numerous things, including different AppleALC layouts, cables, changing volume, removing AppleALC. Nothing we tried works and, as mentioned previously, the cause is not known.

VisualEhrmanntraut, 2024-08-12:
The machines affected by this issue also have the same issue on Linux 5.15. It seems to be fixed in Linux 6.x. Hopefully that gives us a clue on what's wrong.

VisualEhrmanntraut, 2025-03-02:
I have a an AppleALC replacement in development, and as such this functionality will be moved to it when it gets released, so this issue will probably be moved when that time comes.

EinVersuchIstEsWert, 2026-10-13:
Same Probleme here:
Codec: ATI R6xx HDMI
GPU AMD Ryzen 9 5900HX
on  Advanced Micro Devices, Inc. [AMD/ATI] Renoir/Cezanne HDMI/DP Audio Controller
Advanced Micro Devices, Inc. [AMD/ATI] Cezanne [Radeon Vega Series / Radeon Vega Mobile Series]

Feel free and ask for Linux, Windows or Mac OS Terminal commands to fix this or improve NootedRed

EinVersuchIstEsWert, 2026-10-14:
@VisualEhrmanntraut 

I know your Bug and can reproduce it. It has nothing to do with sound codec or driver.

If you have more than one hdmi or displayport, only on one plug is sound working. On the other plugs you will have no sound. In my case i have 2 dp and 2 hdmi. When i change the plug to another dp I got sound.

VisualEhrmanntraut, 2026-10-14:
My laptop has 1 HDMI port and its internal eDP screen, yet it still is broken. How do you explain that, then?

EinVersuchIstEsWert, 2026-10-14:
Ok its another bug. Hmm I described it in a wrong way for my system. One hdmi and one dp socket have sound the other ones not. But all Displays works fine, but I can imagine the problem is the socket in your case case too.

VisualEhrmanntraut, 2026-10-14:
Interesting, as if there's some kind of off-by-one error in the FB or HDA code somehow.

EinVersuchIstEsWert, 2026-10-14:
On linux or windows, sound works on every socket.

VisualEhrmanntraut, 2026-10-14:
No shit, I'm aware.

MarcoH-00, 2026-11-12:
is this already fixed? i have the same problem.

amd ryzen 5 3500u, vega 8

vithalchowdary, 2026-11-15:
@VisualEhrmanntraut 
Fixed the issue. I have written entire thing in the discussions under hdmi audio fix as pull requests are closed. please check the changes and update them for future updates. thanks for letting me fork. https://github.com/vithalchowdary/NootedRed