ChefKissInc/NootedRed #227: Small compute activities or idle hang the system

milovanpms, 2024-02-09:
### macOS Version

Ventura

### What is your CPU's model?

AMD Ryzen 7 5800H

### Please describe the behaviour in detail.

Hello. I'm in the same situation to this issue (https://github.com/ChefKissInc/NootedRed/issues/137), just after entering my session, the system is hanging whether or not I'm doing or launching any program. Tried everything that was on the issue I mentioned before, no changes.

Specs:

Ryzen 7 5800H (Vega 8 Mobile)
OpenCore v0.9.8 (latest release)
NootedRed (latest version)

### What should've happened instead?

No hanging ?

### Bug report submission terms

- [X] I am using the latest version of NootedRed.
- [X] I am on the latest minor and patch version of macOS (ex. 7 is the minor and 10 is the patch of macOS 11.7.10).
- [X] The bug only happens with NootedRed enabled.
- [X] I have ruled out any external factors that could interfere, including, but not limited to, using OCLP or WhateverGreen.

VisualEhrmanntraut, 2024-02-09:
Ok. Why would you create another issue if there is already an issue  submitted for this then? I have a job, I can't be skipping things in my own life just to work on some Hackintosh tomfoolery for essentially free. Just be patient.

VisualEhrmanntraut, 2024-02-09:
My bad, I assumed the other issue was still open. I am too tired. Sorry.

milovanpms, 2024-02-25:
Had a corrupted NootedRed.kext file, just run a release and it's working.

closed 2024-02-25 by milovanpms