ChefKissInc/NootedRed #232: 5600G black screen

szmarczak, 2024-02-29:
### macOS Version

Ventura

### What is your CPU's model?

AMD Ryzen 5 5600G

### Please describe the behaviour in detail.

Boots into black screen. With WhateverGreen it boots to GUI. Using HDMI.

Tried:
- `MacBookPro16,3` SMBIOS
- `iMac20,1` SMBIOS
- `iMacPro1,1` SMBIOS
- disabling `AmdTscSync.kext`

Motherboard: [GIGABYTE B450M-K](https://www.gigabyte.com/Motherboard/B450M-K-rev-10/sp#sp)

### What should've happened instead?

GUI should pop up. Interestingly in the case of #121 it worked, but in this case it does not...

### If applicable, attach the .gpuRestart, .panic, etc file related to this issue.

[config.plist](https://github.com/ChefKissInc/NootedRed/files/14447139/config.txt)

### Bug report submission terms

- [X] I am using the latest version of NootedRed.
- [X] I am on the latest minor and patch version of macOS (ex. 7 is the minor and 10 is the patch of macOS 11.7.10).
- [X] The bug only happens with NootedRed enabled.
- [X] I have ruled out any external factors that could interfere, including, but not limited to, using OCLP or WhateverGreen.

long-zuyang, 2024-03-03:
使用远程桌面工具关闭高动态范围(HDR)就好,我在安装的时候也遇到了这个问题

VisualEhrmanntraut, 2024-03-25:
Timed out waiting for response.

closed 2024-03-25 by VisualEhrmanntraut

szmarczak, 2024-03-26:
Sorry for delay. I just tested 197cfc71465ec87e635f4ebc2ed3d353c4212be8 and it works like a charm! No `-NRedDPDelay` needed. I'm using HDMI so idk if that's why.

I even managed to run iPhone Simulator. It's *a bit* slower than with dGPU (5600 XT) but definitely usable! It lags AF when dragging the current app using the bottom bar upwards (manage open apps view, Android's square button equivalent).