ChefKissInc/NootedRed #235: macOS Sonoma, gpuRestart on the login screen or in System Preferences/Wallpaper

darthsian, 2024-03-13:
### macOS Version

Sonoma

### What is your CPU's model?

AMD Ryzen 7 5700U

### Please describe the behaviour in detail.

There is a beach ball on the login screen for about 20 sec (gpuRestart) and I can't log in. Similar behavior is in System Preferences/Wallpaper, when loading Pictures there is a macOS lag and in the Console you can see gpuRestart several times. If only Color is selected as wallpaper, there is no problem on the login screen.

### What should've happened instead?

Not restarting GPU

### If applicable, attach the .gpuRestart, .panic, etc file related to this issue.

[gpuRestart.zip](https://github.com/ChefKissInc/NootedRed/files/14594648/gpuRestart.zip)

### Bug report submission terms

- [X] I am using the latest version of NootedRed.
- [X] I am on the latest minor and patch version of macOS (ex. 7 is the minor and 10 is the patch of macOS 11.7.10).
- [X] The bug only happens with NootedRed enabled.
- [X] I have ruled out any external factors that could interfere, including, but not limited to, using OCLP or WhateverGreen.

darthsian, 2024-03-14:
So the gpu restart on the login screen occurs even when Wallpaper is selected with Color only. I attach the gpuRestart that occurs in this case.
[Kernel_2024-03-14-005616_SPARKnano.gpuRestart.zip](https://github.com/ChefKissInc/NootedRed/files/14594941/Kernel_2024-03-14-005616_SPARKnano.gpuRestart.zip)

darthsian, 2024-03-14:
After commit 274ba77.
gpuRestart after logging in (this is new one)
[After-logging-in-Kernel_2024-03-14-112858_SPARKnano.gpuRestart.zip](https://github.com/ChefKissInc/NootedRed/files/14600920/After-logging-in-Kernel_2024-03-14-112858_SPARKnano.gpuRestart.zip)

gpuRestart System Settings/Wallpaper
[Wallpaper-Kernel_2024-03-14-112622_SPARKnano.gpuRestart.zip](https://github.com/ChefKissInc/NootedRed/files/14600927/Wallpaper-Kernel_2024-03-14-112622_SPARKnano.gpuRestart.zip)

gpuRestart AvatarPicker (selecting avatar in System Setting/Users & Groups or selecting avatar in System Setting/Apple ID
[Avatar-picker-Kernel_2024-03-14-112741-1_SPARKnano.gpuRestart.zip](https://github.com/ChefKissInc/NootedRed/files/14600944/Avatar-picker-Kernel_2024-03-14-112741-1_SPARKnano.gpuRestart.zip)

darthsian, 2024-03-14:
NootedRed is really having quite a bit of trouble on Sonoma. I've discovered other problems in iMovie where freezing and gpurestart occur. Every now and then the gpu restarts in Safari as well, for example when I try to open a video as “Enter Picture-in-Picture.” in YouTube.
[iMovie-Kernel_2024-03-14-214925_SPARKnano.gpuRestart.zip](https://github.com/ChefKissInc/NootedRed/files/14607894/iMovie-Kernel_2024-03-14-214925_SPARKnano.gpuRestart.zip)
[Safari-Kernel_2024-03-14-220325_SPARKnano.gpuRestart.zip](https://github.com/ChefKissInc/NootedRed/files/14607895/Safari-Kernel_2024-03-14-220325_SPARKnano.gpuRestart.zip)

darthsian, 2024-03-15:
I forgot to mention that these crashes don't happen on Ventura.

darthsian, 2024-04-01:
> Has there been an improvement on the latest commit?

Commit 8967a8d.
Unfortunately, still the same compute crashes.

idoan, 2024-06-15:
> Yes it as less crashes. Thanks for your work

Do you guys still have occasional crashes due to Sonoma's incompability with nootedred, or is it thoroughly fixed by now?

Rix565, 2024-06-19:
Yes, same problems all over again

VisualEhrmanntraut, 2024-08-13:
I have some ideas but because of Lilu's limitations I can't quite implement them yet.

darthsian, 2024-09-20:
Hi, after a quick test of Sequoia I can confirm that this issue is still present. I will do more extensive testing later.

MiniPC, Ryzen 7 5800H, macOS 15.0

VisualEhrmanntraut, 2024-09-20:
Thanks for testing. I will update the issue accordingly

darthsian, 2024-09-22:
gpuRestart VTDecoderXPCServ
[Kernel_2024-09-22-184117.gpuRestart.zip](https://github.com/user-attachments/files/17090465/Kernel_2024-09-22-184117.gpuRestart.zip)

darthsian, 2024-09-22:
> This happens randomly on app store when a movie fails to play. Do you have any kernel panic log related to VTDecoderXPCServ ? What happens if you use process monitor to kill VTDecoderXPCServ ?

I haven't had a Kernel Panic related to VTDecoderXPCServ yet. Only gpuRestart.

epokuser, 2024-10-04:
hi I see opening safari and doing video playback then open new tab causes crash too!

darthsian, 2024-10-05:
> hi I see opening safari and doing video playback on youtube then open new tab causes crash too!

Yes, I can confirm that.
These crashes are indeed very bothersome and occur in many other operations. I hope that @VisualEhrmanntraut  can resolve this in the future. As it is, macOS from Sonoma above is very hard to use.

VisualEhrmanntraut, 2024-10-05:
@darthsian Await https://github.com/orgs/ChefKissInc/discussions/58.

darthsian, 2024-10-05:
> @darthsian Await https://github.com/orgs/ChefKissInc/discussions/58.

@VisualEhrmanntraut Yes, an excellent project. I wish you a lot of strength, because it’s incredible that you are managing all of this on your own.

moishiteru, 2024-10-19:
so we need to wait for a newer version of nootedred.kext or the artic tosh project?

nononymousse, 2024-10-19:
i see that they are new commits on NootedRed so i guess both are been worked on

VisualEhrmanntraut, 2024-10-19:
yeah I wouldn't just leave the projects in a buggy state

nononymousse, 2024-10-19:
thanks a lot, can’t wait to see the first release of Arcintosh ^^

VisualEhrmanntraut, 2024-10-27:
No, I need these logs to be able to diagnose GPU crashes.

Win10Q, 2024-11-06:
Macos15sequoia on raven when i open the settings it has panic
this is my panic log
mp_kdp_enter() timed-out on cpu 2, NMI-ing
mp_kdp_enter() NMI pending on cpus: 0 1 3 4 5 6 7
mp_kdp_enter() timed-out during locked wait after NMI;expected 8 acks but received 1 after 163083 loops in 244013819 ticks
panic(cpu 2 caller 0xffffff801af53db5): mp_rv_action() entry: 6 of 8 responses, start: 0xdb27e103db, cur: 0xdb620e70f4 @mp.c:902
Panicked task 0xffffff8b666ff4c0: 3 threads: pid 524: MTLCompilerServi
Backtrace (CPU 2), panicked thread: 0xffffff94fdf600e0, Frame : Return Address
0xffffffca3bdbdab0 : 0xffffff801adebc91 
0xffffffca3bdbdb00 : 0xffffff801af58a0c 
0xffffffca3bdbdb40 : 0xffffff801af482b3 
0xffffffca3bdbdc10 : 0xffffff801ad81971 
0xffffffca3bdbdc30 : 0xffffff801adebf87 
0xffffffca3bdbdd30 : 0xffffff801adeb622 
0xffffffca3bdbdda0 : 0xffffff801b5d4078 
0xffffffca3bdbde90 : 0xffffff801af53db5 
0xffffffca3bdbdec0 : 0xffffff801af53122 
0xffffffca3bdbdf20 : 0xffffff801af52c96 
0xffffffca3bdbdf60 : 0xffffff801af51c6a 
0xffffffca3bdbdf80 : 0xffffff801af47845 
0xffffffca3bdbdfe0 : 0xffffff801ad81a9a 
0xffffffbe0f76ffa0 : 0xffffff801ad8189f 

Process name corresponding to current thread (0xffffff94fdf600e0): MTLCompilerService
Boot args: -v npci=0x3000 revpatch=sbvmm,cpuname,auto darkwake=no brcmfx-country=CN 

Mac OS version:
24B83

Kernel version:
Darwin Kernel Version 24.1.0: Thu Oct 10 21:02:27 PDT 2024; root:xnu-11215.41.3~2/RELEASE_X86_64
Kernel UUID: 15F8E5B4-41DF-31C2-8CF5-F486066C59B9
roots installed: 0
KernelCache slide: 0x000000001aa00000
KernelCache base:  0xffffff801ac00000
Kernel slide:      0x000000001aae4000
Kernel text base:  0xffffff801ace4000
__HIB  text base: 0xffffff801ab00000
System model name: MacBookPro16,3 (Mac-E7203C0F68AA0004)
System shutdown begun: NO
Panic diags file available: YES (0x0)
Hibernation exit count: 0

System uptime in nanoseconds: 435394499890
Last Sleep:           absolute           base_tsc          base_nano
  Uptime  : 0x000000655f896021
  Sleep   : 0x0000000000000000 0x0000000000000000 0x0000000000000000
  Wake    : 0x0000000000000000 0x00000007033932c9 0x0000000000000000
Compressor Info: 0% of compressed pages limit (OK) and 0% of segments limit (OK) with 0 swapfiles and OK swap space
Zone info:
  Zone map: 0xffffff8031144000 - 0xffffffa031144000
  . PGZ   : 0xffffff8031144000 - 0xffffff8033945000
  . VM    : 0xffffff8033945000 - 0xffffff8500011000
  . RO    : 0xffffff8500011000 - 0xffffff86997aa000
  . GEN0  : 0xffffff86997aa000 - 0xffffff8b65e76000
  . GEN1  : 0xffffff8b65e76000 - 0xffffff9032543000
  . GEN2  : 0xffffff9032543000 - 0xffffff94fec10000
  . GEN3  : 0xffffff94fec10000 - 0xffffff99cb2dd000
  . DATA  : 0xffffff99cb2dd000 - 0xffffffa031144000
  Metadata: 0xffffffa051954000 - 0xffffffa071954000
  Bitmaps : 0xffffffa071954000 - 0xffffffa075554000
  Extra   : 0 - 0

RijoRaj, 2024-12-24:
Sir, I am new to this place and I came here as I was also facing the same issue of freezing after few seconds of logging in. I can see that you have removed the tag Bug, last month, does this mean the issue is fixed ?. And if so please mention how, since I have used the latest artifact avaliable and still it freezes in Sonoma And Sequoia.

VisualEhrmanntraut, 2025-03-22:
Is this still happening with latest commit?

Win10Q, 2025-03-22:
I try it on my 3500u but the gpu doesn‘t work

[image]

VisualEhrmanntraut, 2025-03-22:
@Win10Q Your Lilu is just out of date.

Win10Q, 2025-03-22:
oh，my false

Win10Q, 2025-03-22:
i update lilu and gpu worked verygood，but I clicked on bilibili to watch a video, and then the system became very stuck, as before.

Win10Q, 2025-03-22:
I tried to change other wallpapers, but they still get stuck as before.😟

Win10Q, 2025-03-23:
mp_kdp_enter() timed-out on cpu 2, NMI-ing
mp_kdp_enter() NMI pending on cpus: 0 1 3 4 5 6 7
mp_kdp_enter() timed-out during locked wait after NMI;expected 8 acks but received 1 after 363115 loops in 244013990 ticks
panic(cpu 2 caller 0xffffff800b553db5): mp_rv_action() entry: 4 of 8 responses, start: 0x87f4f83dd4, cur: 0x882f25acd0 @mp.c:902
Panicked task 0xffffff9e75c4d2e8: 215 threads: pid 0: kernel_task
Backtrace (CPU 2), panicked thread: 0xffffff9e75a42b40, Frame : Return Address
0xffffff89a3eddab0 : 0xffffff800b3ebc91 
0xffffff89a3eddb00 : 0xffffff800b558a0c 
0xffffff89a3eddb40 : 0xffffff800b5482b3 
0xffffff89a3eddc10 : 0xffffff800b381971 
0xffffff89a3eddc30 : 0xffffff800b3ebf87 
0xffffff89a3eddd30 : 0xffffff800b3eb622 
0xffffff89a3eddda0 : 0xffffff800bbd4078 
0xffffff89a3edde90 : 0xffffff800b553db5 
0xffffff89a3eddec0 : 0xffffff800b553122 
0xffffff89a3eddf20 : 0xffffff800b552c96 
0xffffff89a3eddf60 : 0xffffff800b551c6a 
0xffffff89a3eddf80 : 0xffffff800b547845 
0xffffff89a3eddfe0 : 0xffffff800b381a9a 
0xfffffffd72f13bf0 : 0xffffff800b52825d 
0xfffffffd72f13cb0 : 0xffffff800b528e81 
0xfffffffd72f13d00 : 0xffffff800b4b0bbb 
0xfffffffd72f13d80 : 0xffffff800ef26641 
0xfffffffd72f13dd0 : 0xffffff800ef1e8f3 
0xfffffffd72f13e70 : 0xffffff800cb8c769 
0xfffffffd72f13e90 : 0xffffff800cb9187c 
0xfffffffd72f13ed0 : 0xffffff800baf8b6a 
0xfffffffd72f13f20 : 0xffffff800baf72de 
0xfffffffd72f13f60 : 0xffffff800baf6916 
0xfffffffd72f13fa0 : 0xffffff800b38119e 
      Kernel Extensions in backtrace:
         com.apple.driver.AppleSMC(3.1.9)[4364B20F-0128-324C-AC42-FC3382C9784A]@0xffffff800cb84000->0xffffff800cb9dfff
            dependency: com.apple.driver.watchdog(1)[E2B50980-FE99-33C7-8EBF-4D0F2C67FBF7]@0xffffff800e22d000->0xffffff800e22ffff
            dependency: com.apple.iokit.IOACPIFamily(1.4)[5CCDA23F-974B-36FE-991B-F492F97C7C90]@0xffffff800d3a9000->0xffffff800d3aafff
            dependency: com.apple.iokit.IOPCIFamily(2.9)[356D7C71-CB4B-37B3-9AB2-7160A434CC2A]@0xffffff800d823000->0xffffff800d856fff
         as.vit9696.VirtualSMC(1.3.4)[84598C28-7429-3F87-9318-65E2C76F7C44]@0xffffff800ef1e000->0xffffff800ef33fff
            dependency: as.vit9696.Lilu(1.7.0)[276C35BC-B576-3049-AF11-4E11E8EBE459]@0xffffff800eeef000->0xffffff800ef1afff
            dependency: com.apple.iokit.IOACPIFamily(1.4)[5CCDA23F-974B-36FE-991B-F492F97C7C90]@0xffffff800d3a9000->0xffffff800d3aafff

Process name corresponding to current thread (0xffffff9e75a42b40): kernel_task
Boot args: npci=0x3000 revpatch=sbvmm,cpuname darkwake=no brcmfx-country=#a debug=0x144 

Mac OS version:
24B83

Kernel version:
Darwin Kernel Version 24.1.0: Thu Oct 10 21:02:27 PDT 2024; root:xnu-11215.41.3~2/RELEASE_X86_64
Kernel UUID: 15F8E5B4-41DF-31C2-8CF5-F486066C59B9
roots installed: 0
KernelCache slide: 0x000000000b000000
KernelCache base:  0xffffff800b200000
Kernel slide:      0x000000000b0e4000
Kernel text base:  0xffffff800b2e4000
__HIB  text base: 0xffffff800b100000
System model name: MacBookPro16,3 (Mac-E7203C0F68AA0004)
System shutdown begun: NO
Panic diags file available: YES (0x0)
Hibernation exit count: 0

System uptime in nanoseconds: 260728902544
Last Sleep:           absolute           base_tsc          base_nano
  Uptime  : 0x0000003cb4a79668
  Sleep   : 0x0000000000000000 0x0000000000000000 0x0000000000000000
  Wake    : 0x0000000000000000 0x000000090e224c98 0x0000000000000000
Compressor Info: 0% of compressed pages limit (OK) and 0% of segments limit (OK) with 0 swapfiles and OK swap space
Zone info:
  Zone map: 0xffffff9340ddd000 - 0xffffffb340ddd000
  . PGZ   : 0xffffff9340ddd000 - 0xffffff93435de000
  . VM    : 0xffffff93435de000 - 0xffffff980fcaa000
  . RO    : 0xffffff980fcaa000 - 0xffffff99a9443000
  . GEN0  : 0xffffff99a9443000 - 0xffffff9e75b0f000
  . GEN1  : 0xffffff9e75b0f000 - 0xffffffa3421dc000
  . GEN2  : 0xffffffa3421dc000 - 0xffffffa80e8a9000
  . GEN3  : 0xffffffa80e8a9000 - 0xffffffacdaf76000
  . DATA  : 0xffffffacdaf76000 - 0xffffffb340ddd000
  Metadata: 0xffffffb381571000 - 0xffffffb3a1571000
  Bitmaps : 0xffffffb3a1571000 - 0xffffffb3a5171000
  Extra   : 0 - 0

this is my kernel panic log

VisualEhrmanntraut, 2025-03-23:
I must admit that this kernel panic log is useless because you are lacking `keepsyms=1`, also I notice you have the about this mac patch enabled, you should read https://github.com/orgs/ChefKissInc/discussions/64

Win10Q, 2025-03-23:
mp_kdp_enter() timed-out on cpu 2, NMI-ing
mp_kdp_enter() NMI pending on cpus: 0 1 3 4 5 6 7
mp_kdp_enter() timed-out during locked wait after NMI;expected 8 acks but received 1 after 413332 loops in 244013504 ticks
panic(cpu 2 caller 0xffffff8016f53db5): mp_rv_action() entry: 4 of 8 responses, start: 0x4557b8a112, cur: 0x4591e608ac @mp.c:902
Panicked task 0xffffff9089490870: 1 threads: pid 739: promotedcontentd
Backtrace (CPU 2), panicked thread: 0xffffff9086b280e0, Frame : Return Address
0xffffffc7bbe73ab0 : 0xffffff8016debc91 mach_kernel : _handle_debugger_trap + 0x4c1
0xffffffc7bbe73b00 : 0xffffff8016f58a0c mach_kernel : _kdp_i386_trap + 0x11c
0xffffffc7bbe73b40 : 0xffffff8016f482b3 mach_kernel : _kernel_trap + 0x763
0xffffffc7bbe73c10 : 0xffffff8016d81971 mach_kernel : _return_from_trap + 0xc1
0xffffffc7bbe73c30 : 0xffffff8016debf87 mach_kernel : _DebuggerTrapWithState + 0x67
0xffffffc7bbe73d30 : 0xffffff8016deb622 mach_kernel : _panic_trap_to_debugger + 0x1e2
0xffffffc7bbe73da0 : 0xffffff80175d4078 mach_kernel : _panic + 0x81
0xffffffc7bbe73e90 : 0xffffff8016f53db5 mach_kernel : _mp_safe_spin_lock + 0x3b5
0xffffffc7bbe73ec0 : 0xffffff8016f53122 mach_kernel : _mp_cpus_call_cpu_init + 0x452
0xffffffc7bbe73f20 : 0xffffff8016f52c96 mach_kernel : _cpu_signal_handler + 0x1f6
0xffffffc7bbe73f60 : 0xffffff8016f51c6a mach_kernel : _lapic_interrupt + 0x4a
0xffffffc7bbe73f80 : 0xffffff8016f47845 mach_kernel : _interrupt + 0xe5
0xffffffc7bbe73fe0 : 0xffffff8016d81a9a mach_kernel : _hndl_allintrs + 0x9a

Process name corresponding to current thread (0xffffff9086b280e0): promotedcontentd
Boot args: npci=0x3000 revpatch=sbvmm,cpuname darkwake=no brcmfx-country=#a debug=0x100 keepsyms=1 -v 

Mac OS version:
24B83

Kernel version:
Darwin Kernel Version 24.1.0: Thu Oct 10 21:02:27 PDT 2024; root:xnu-11215.41.3~2/RELEASE_X86_64
Kernel UUID: 15F8E5B4-41DF-31C2-8CF5-F486066C59B9
roots installed: 0
KernelCache slide: 0x0000000016a00000
KernelCache base:  0xffffff8016c00000
Kernel slide:      0x0000000016ae4000
Kernel text base:  0xffffff8016ce4000
__HIB  text base: 0xffffff8016b00000
System model name: MacBookPro16,3 (Mac-E7203C0F68AA0004)
System shutdown begun: NO
Panic diags file available: YES (0x0)
Hibernation exit count: 0

System uptime in nanoseconds: 129748540589
Last Sleep:           absolute           base_tsc          base_nano
  Uptime  : 0x0000001e359dd40d
  Sleep   : 0x0000000000000000 0x0000000000000000 0x0000000000000000
  Wake    : 0x0000000000000000 0x000000065dc2dd89 0x0000000000000000
Compressor Info: 0% of compressed pages limit (OK) and 0% of segments limit (OK) with 0 swapfiles and OK swap space
Zone info:
  Zone map: 0xffffff8086031000 - 0xffffffa086031000
  . PGZ   : 0xffffff8086031000 - 0xffffff8088832000
  . VM    : 0xffffff8088832000 - 0xffffff8554efe000
  . RO    : 0xffffff8554efe000 - 0xffffff86ee697000
  . GEN0  : 0xffffff86ee697000 - 0xffffff8bbad63000
  . GEN1  : 0xffffff8bbad63000 - 0xffffff9087430000
  . GEN2  : 0xffffff9087430000 - 0xffffff9553afd000
  . GEN3  : 0xffffff9553afd000 - 0xffffff9a201ca000
  . DATA  : 0xffffff9a201ca000 - 0xffffffa086031000
  Metadata: 0xffffffbe06f85000 - 0xffffffbe26f85000
  Bitmaps : 0xffffffbe26f85000 - 0xffffffbe2ab85000
  Extra   : 0 - 0

Thanks for the guidance, I added keepsyms=1 and then retriggered the stuck, hope this time the logs are useful

paulohenriquefurtadopereira, 2025-12-29:
> This issue happens a lot with Ryzen 3 3200G on Sonoma and above. The whole system crashes within the first minute of logging in with nothing open. Note: I have set a still wallpaper and dont have any memoji as my user image. Afaik, other 3200G users have also faced this similar issue. Maybe it's just some problem with this chip because things seem to run more or less fine on other APUs.
> 
> Just wanted to add this info as it has not been mentioned in this thread before, otherwise great work :)

> Esse problema ocorre com frequência com o Ryzen 3 3200G no Sonoma e versões superiores. O sistema trava completamente no primeiro minuto após o login, mesmo sem nenhum programa aberto. Observação: configurei um papel de parede estático e não tenho nenhum memoji como imagem de usuário. Pelo que sei, outros usuários do 3200G também enfrentaram esse problema semelhante. Talvez seja um problema específico desse chip, já que tudo parece funcionar mais ou menos bem em outras APUs.
> 
> Só queria acrescentar esta informação, pois ainda não foi mencionada neste tópico. De resto, ótimo trabalho :)

The same thing happens here with the Vega 3 2200g. I managed to grow the Sequoia and the Tahoe, but both freeze completely after logging in, in less than 2 minutes.

hoaug-tran, 2026-01-02:
I’m experiencing lag and slowness (no crashes; interactions are just very slow and the mouse cursor keeps spinning) when I open the App Store and Apple Music (if I quickly switch to a playlist, the issue goes away).

I’ve already set the wallpaper to a solid color, including the Lock Screen, and I’m also using only solid images for the avatar. Is there any way to use the App Store normally?

I’m using the latest kexts: ForgedInvariant and NootedRed (I was try to remove ForgedInvariant and switched to a fork of CPUTscSync for AMD).

EDIT: I found that enabling Reduce Transparency in System Settings → Accessibility makes the App Store and Apple Music no longer slow or laggy.

VisualEhrmanntraut, 2026-05-23:
However, if anyone else has any ideas, feel free to chime in. (the account who was blocked a few comments ago is because the comment was hallucinated by an LLM instead of a real response from an experienced developer)

BlkDev9, 2026-05-24:
Would the changes between beta 4 and 14.0 release candidate be too many to find the difference we’re looking for?

BlkDev9, 2026-05-24:
After testing beta 4 a while I noticed that even though compute heavy functions seem to work the system is still really unstable, windowserver crashes all the time, and now I wonder whether this is the beta’s fault or the changes already began getting implemented here

BlkDev9, 2026-05-24:
Okay so there’s one problem with NR, I switched back to latest build and now after a couple minutes after boot, the system freezes for 10 seconds and then it turns responsive again… only problem is that it stutters like crazy after that, it’s like it renders a frame every 2 seconds until I reboot, 2023 build doesn’t have this issue at all, seems to have been just as stable as Ventura the last 40 minutes, I wonder what NR change would cause something like this

Edit: it’s definitely not as unstable as the issues risen from beta 4 where it straight up crashes after loading the wallpaper

VisualEhrmanntraut, 2026-05-25:
The test was meant for beta 1 and beta 3, as the QC tester reported to me that beta 1 stopped stuttering with the before v2 one.

VisualEhrmanntraut, 2026-05-26:
[image]
dafuq?

BlkDev9, 2026-05-26:
Strange issue with the newer commits, ~system still doesn’t crash~ and runs mostly smooth but it’s stuttering quite frequently, Tahoe even crashed, I just checked to make sure gfxoff was still disabled in my fork

Edit: Sonoma just freezed and crashed :(

VisualEhrmanntraut, 2026-05-26:
I guess there's only one way to find out, by trying the fix ourselves, but on all apus

VisualEhrmanntraut, 2026-05-28:
Found what VTMTSComputeFunction means. It means "VideoToolbox Metal Transfer Session Compute Function". The binary is full of outlined and ultra-optimised, unsymbolicated code so I'm still not sure if there's alternative paths for it.

VisualEhrmanntraut, 2026-05-28:
```
defaults write com.apple.coremedia allowMetalTransferSession -bool NO
```
what happens with that?

tilt-rgb, 2026-05-30:
* Vídeo hardware (gpu) working here, e.e, with this workaround:

[image]

sudo defaults write /Library/Preferences/com.apple.coremedia allowMetalTransferSession -bool NO

sudo chmod 644 /Library/Preferences/com.apple.coremedia.plist

Yes, this fixes hardware decoding here, maybe strange? I sending a video showing: https://youtu.be/kcyUjslPBl8

* YouTube only shows 2160p if gpu is compatible with 2160p, u maybe known this...

More proofs: call my in telegram if u need: https://t.me/k_a_s_a_n_e_t_e_t_o

zackmsa777-a11y, 2026-11-05:
Same issue here on an Acer Nitro 5 AN515-44 with a Ryzen 5 4600H. With NootedRed 0.8.10 enabled, the internal screen goes black after verbose boot. With NootedRed disabled, I can reach macOS Sequoia 15.4.1 Recovery on the same screen.