ChefKissInc/NootedRed #236: Brightness control on Vega 8 (Raven Ridge) broken after updating to latest version Nootedred

MizuKumi, 2024-03-14:
### macOS Version

Ventura

### What is your CPU's model?

AMD Ryzen 5 PRO 3500U

### Please describe the behaviour in detail.

After upgrading to the latest version of NootedRed: _https://github.com/ChefKissInc/NootedRed/actions/runs/8277805359_ **(commit [274ba77])**

The brightness control stopped working properly, the brightness reduction via the native "Brightness Slider" from macOS only goes down one point from the maximum brightness value, the brightness does not go any lower after moving the slider.

In the previous version of Nootedred:
_https://github.com/ChefKissInc/NootedRed/actions/runs/7765787829_ **(Commit [eff598f])**

The brightness control was working properly. Via the native "Brightness Slider" from macOS.

Videos demonstrating the problem are attached in two files.

Before:
https://github.com/ChefKissInc/NootedRed/assets/116596455/5e0ec249-8dbe-4360-8178-868efe9eb4d6

After:
https://github.com/ChefKissInc/NootedRed/assets/116596455/285fd8c8-38a0-4218-a8a4-97404bcb6f7d

### What should've happened instead?

Brightness control, well, what else is there? :D

### If applicable, attach the .gpuRestart, .panic, etc file related to this issue.

Pointless, the problem is inside Nootedred, apparently this problem appeared after **commit d6554a3:** _https://github.com/ChefKissInc/NootedRed/actions/runs/8253163850_

### Bug report submission terms

- [X] I am using the latest version of NootedRed.
- [X] I am on the latest minor and patch version of macOS (ex. 7 is the minor and 10 is the patch of macOS 11.7.10).
- [X] The bug only happens with NootedRed enabled.
- [X] I have ruled out any external factors that could interfere, including, but not limited to, using OCLP or WhateverGreen.

VisualEhrmanntraut, 2024-03-14:
What is the display type?

MizuKumi, 2024-03-14:
Where can I see this in macOS?

VisualEhrmanntraut, 2024-03-14:
I meant LCD, OLED, ... if it has HDR, etc.

MizuKumi, 2024-03-14:
[image]

VisualEhrmanntraut, 2024-03-16:
It seems like this issue is specific to Raven (DCN 1). I have not seen the issue occur on Renoir (DCN 2). It is possible that AMDRadeonX6000Framebuffer has a bug with the DMCU control. I will investigate how to solve this without removing the newly added Catalina support for brightness control and etc.

MizuKumi, 2024-03-16:
Okay. I look forward to your correction. And yes, thanks for such a great project, you made it possible for me not to buy a macbook for overprice. :)

MacintoshSSD, 2024-06-28:
An idea：maybe we can design a command for users to add boot-args to disable Catalina brightness control code, or delete Catalina brightness control code, because Catalina is outdated.

VisualEhrmanntraut, 2024-08-19:
I improved the situation a little bit.

commit 0a0cdc45 (2024-10-08): fix: Use old PWM brightness control logic on Raven

Closes #236
Temporary workaround. FIXME!