ChefKissInc/NootedRed #241: Twitter/X makes the OS unresponsive when trying to type 2FA

PGBSean, 2024-03-29:
### macOS Version

Sonoma

### What is your CPU's model?

AMD Ryzen 5 3400G

### Please describe the behaviour in detail.

I clicked "Sign in to Google" on Twitter/X. When prompt for 2FA code, the OS start to hang up.

### What should've happened instead?

Twitter/X will still can go through the sign-in process effortlessly without any issue to the system.

### If applicable, attach the .gpuRestart, .panic, etc file related to this issue.

[Logs.zip](https://github.com/ChefKissInc/NootedRed/files/14804290/Logs.zip)

### Bug report submission terms

- [X] I am using the latest version of NootedRed.
- [X] I am on the latest minor and patch version of macOS (ex. 7 is the minor and 10 is the patch of macOS 11.7.10).
- [X] The bug only happens with NootedRed enabled.
- [X] I have ruled out any external factors that could interfere, including, but not limited to, using OCLP or WhateverGreen.

VisualEhrmanntraut, 2024-03-29:
Arc is an Open-GL app, which means this is...

VisualEhrmanntraut, 2024-03-29:
Duplicate of #158

closed 2024-03-29 by VisualEhrmanntraut