ChefKissInc/NootedRed #243: Rendering issues possible caused by nootedred

rishthedev, 2024-03-30:
### macOS Version

Sonoma

### What is your CPU's model?

Ryzen 7 5700G 

### Please describe the behaviour in detail.

[image]
Randomly a red rectangle appears that covers the whole of safari. It will dissappear after i reload but may have the issue again. 

### What should've happened instead?

No render issues on safari 

### Bug report submission terms

- [X] I am using the latest version of NootedRed.
- [X] I am on the latest minor and patch version of macOS (ex. 7 is the minor and 10 is the patch of macOS 11.7.10).
- [X] The bug only happens with NootedRed enabled.
- [X] I have ruled out any external factors that could interfere, including, but not limited to, using OCLP or WhateverGreen.

jpsaturnino, 2024-04-06:
This can be caused by insufficient VRAM size, look at https://chefkissinc.github.io/nred#prerequisites. You can do this by updating your bios, search by `UMA mode` and force to at least 1gb depending on your available memory.

[image]

Dummy-Soft, 2024-04-16:
Disabel hardware acceleration in the browser

VisualEhrmanntraut, 2024-04-19:
Here are the most common things to look out for:
- Outdated version of this kext.
- Conflicts with other kexts (e.g. WhateverGreen).
- Issues caused by other kexts like memory corruption (e.g. AirportItlwm).
- Using cheap hacks like kext/library downgrades (e.g. OCLP, BFixup).
- Duplicate kexts.
- Malformed config.plist caused by configurator software (OCAT, OCC, etc).
- Lack of TSC synchronisation (use AmdTscSync).
- Badly configured TSC synchronisation (set IOCPUNumber in the Info.plist to number of threads minus one for macOS 11 and older, 0 for anything newer).
- Outdated macOS minor version (e.g. macOS 11.0.0 instead of latest).
- Using beta versions of macOS.

VisualEhrmanntraut, 2024-05-29:
Timed out waiting for response.

closed 2024-05-29 by VisualEhrmanntraut