ChefKissInc/NootedRed #244: Lenovo Yoga-14S-2020ARE, CPU4800U, mouse stuck, automatic restart, yellow screen!

yuanxun9, 2024-04-01:
### macOS Version

Sonoma

### What is your CPU's model?

4800U

### Please describe the behaviour in detail.

Hello, my laptop is Lenovo Yoga-14S-2020ARE, with a CPU of 4800U. It uses the latest version of the graphics card driver CI # 1090: 8019036 NootedRed.kext, but the mouse is stuck and automatically restarts. Uncheck NootedRed.kext in the OC configuration file to boot up normally without a driver kernel! Could you please check the log if you have time? Thank you for your work!!

### What should've happened instead?

Mouse stuck, automatic restart, yellow screen!

### Bug report submission terms

- [X] I am using the latest version of NootedRed.
- [X] I am on the latest minor and patch version of macOS (ex. 7 is the minor and 10 is the patch of macOS 11.7.10).
- [X] The bug only happens with NootedRed enabled.
- [X] I have ruled out any external factors that could interfere, including, but not limited to, using OCLP or WhateverGreen.

yuanxun9, 2024-04-01:
[image]
[image]
[image]
[opencore.zip](https://github.com/ChefKissInc/NootedRed/files/14822587/opencore.zip)
Thank you for your reply. I'll give it a try! Thank you!!

VisualEhrmanntraut, 2024-04-01:
The OpenCore log is only OpenCore's logs. This is not of use to me.

yuanxun9, 2024-04-01:
> The OpenCore log is only OpenCore's logs. This is not of use to me.

Hello, after unchecking the graphics card driver and entering the system, the wallpaper was changed to PNG format, but the fault still persists! Selected a few days of log files, please take a look at those that are useful! Thank you again, thank you for your hard work!
[Uploading 归档.zip…]()

VisualEhrmanntraut, 2024-04-01:
You clicked comment before the file got a chance to upload.

VisualEhrmanntraut, 2024-04-01:
Actually, are you using a DEBUG or RELEASE build?

yuanxun9, 2024-04-02:
> 实际上，您使用的是DEBUG还是RELEASE构建？

Hello, it's RELEASE! Under NVRAM, 7C436110-AB2A-4BBBB-A880-FE41995C9F82 has changed several parameters - wegnoegpu keepsyms=1- vi2c force pooling debug=0x100 alcid=21- no_compat_check, disabled FeatureUnlock. kext and SMCRadeonGPU. kext are now functioning properly!

VisualEhrmanntraut, 2024-04-02:
Okay, this is definitely not an issue with NootedRed; closing.

closed 2024-04-02 by VisualEhrmanntraut