ChefKissInc/NootedRed #245: Monitor doesn't getting recognized on almost every startup

binxr, 2024-04-06:
### macOS Version

Monterey

### What is your CPU's model?

Ryzen 5 3350G PRO

### Please describe the behaviour in detail.

Sometimes when I boot up my hackintosh my monitor not get recognized, so I can't use it because my screen turns no signal, anyone knows how to make it back to normal?

### What should've happened instead?

My monitor should be detected on startup and showing the Mac OS normally.

### Bug report submission terms

- [X] I am using the latest version of NootedRed.
- [X] I am on the latest minor and patch version of macOS (ex. 7 is the minor and 10 is the patch of macOS 11.7.10).
- [X] The bug only happens with NootedRed enabled.
- [X] I have ruled out any external factors that could interfere, including, but not limited to, using OCLP or WhateverGreen.

binxr, 2024-04-06:
Does anyone here know what's happening? it was annoying to be restarting my computer several times to go back to normal stage, I'm testing this project for a year, and this happened over and over again.

binxr, 2024-04-07:
Someone knows why it crash and have hard artifacts in apps with OpenGL?

VisualEhrmanntraut, 2024-04-08:
Thanks for the very detailed bug report. It is really helpful that you specified things like logs when the issue happens, the connection type, and all the other useful info. I will surely be able to resolve this in no time.

binxr, 2024-04-08:
Thanks for your response! I will add a log file in my next message and we can solve the problem! As for the port I use, I casually use VGA but I don't think that's the problem because when I connect to HDMI the same thing happens

VisualEhrmanntraut, 2024-04-08:
macOS does not support VGA, but I would expect the motherboard to convert the signal internally to DisplayPort

binxr, 2024-04-08:
@VisualEhrmanntraut of course bro, mac convert to displayport, but here's my EFI with my config.plist
if you see something wrong in the config tell me what is going on ok? im newbie!
[EFI.zip](https://github.com/ChefKissInc/NootedRed/files/14901308/EFI.zip)

VisualEhrmanntraut, 2024-04-08:
Configurators (OCAT, etc) are disallowed because they cause various issues.

VisualEhrmanntraut, 2024-04-08:
and AmdTscSync is unconfigured, this is why your system works improperly. Since you're using Monterey, IOCPUNumber needs to be 0

binxr, 2024-04-08:
@VisualEhrmanntraut But if I make an EFI from scratch normally without configurators, I have the same problem that's why I made this bug tag

VisualEhrmanntraut, 2024-04-08:
sir, read all the messages before replying

VisualEhrmanntraut, 2024-04-08:
Is your BIOS up to date? Also, you're using SSDT-PNLF on a desktop.

binxr, 2024-04-08:
@VisualEhrmanntraut Yes, is in the last update but the update is from 2022-05-05

I will remove the ssdt-pnlf if it is not necessary

VisualEhrmanntraut, 2024-04-08:
Yes, you should remove it in case it's attempting to set the brightness or something.

binxr, 2024-04-08:
@VisualEhrmanntraut what do you think the problem is? I have to restart the PC many times for it to work normally and for my screen to have a "signal"

VisualEhrmanntraut, 2024-04-08:
I'm not sure. But you should try removing the PNLF for now. If that doesn't work, I don't see anything else going on in here; there's not much in this EFI. If the issue still happens with NRed disabled, then there's something else going on. If not, then it could be a bug with the firmware (as I have not seen this issue on any other machine) or a bug with Apple's drivers (NRed isn't really a driver but rather just a patcher for Apple's drivers).

binxr, 2024-04-08:
@VisualEhrmanntraut I will restart the system, when it works I will remove the SSDT-PLNF

But I see one thing, when I restart the PC many times in a row, there are times when it opens verbose, it automatically restarts the machine, and there are times when this happens the machine turns on, but they are rare

VisualEhrmanntraut, 2024-04-08:
you also have XhciPortLimit enabled which is broken since 11.3 or something. and it's also not actually needed on AMD

binxr, 2024-04-08:
@VisualEhrmanntraut I disabled XhciPortLimit and SSDT-PLNF, and the screen still has no signal

If I were to restart just once I would be less annoyed, but I have to restart MANY TIMES to be able to access my hackintosh

binxr, 2024-04-08:
I'll send a video

binxr, 2024-04-08:
my problem was not solved,It didn’t even come close to being resolved. But I have another problem and I will open another bug tag, 
IF THERE IS A SOLUTION TO THE PROBLEM WITH THIS TAG, PLEASE COMMENT!

VisualEhrmanntraut, 2024-04-08:
> my problem was not solved,It didn’t even come close to being resolved.

Don't owe you one, this is a hobby project. At this point, it's likely a firmware bug or a bug with Apple's drivers. x86 macOS is soon to be EOL anyway.

binxr, 2024-04-08:
@VisualEhrmanntraut If you want the situation to be like this, fine, I'll try to get more compatible hardware, thank you very much for your time, it was a pleasure trying to make this work, I hope the project goes far and that the people who use it have better results than the man, see you another day mate!

closed 2024-04-08 by binxr

VisualEhrmanntraut, 2024-04-08:
I'm sorry I couldn't help. I've not seen this issue before with NootedRed. I can't do much without the dmesg logs from when your system is in a black screen.

nononymousse, 2024-04-08:
> Don't owe you a solution; this is just a hobby project.

can’t say it better, i have read all the conversation and it’s really infuriating. the dev(s?) don’t own you support, they are literally doing that in their free time when they can. try to remember it next time you make a issue on any repo, peoples have lives

and to the dev, thank you so much for all the free work you are doing. you made wonders by being able to support AMD APU’s and i really hope the interactions you will have in the future will be more pleasant, big support on you

binxr, 2024-04-08:
@VisualEhrmanntraut It's okay, it's not your fault that you can't make something in development work perfectly on any computer, I've been trying to hackintosh this computer for about 3 years, I never really managed to do it, it was a lot of frustration, but I never gave up, now I really have a reason to give up and get new hardware, it was a pleasure to test the project from the beginning and I hope it becomes something 100% functional for AMD APUs!

binxr, 2024-04-08:
> can’t say it better, i have read all the conversation and it’s really infuriating. the dev(s?) don’t own you support, they are literally doing that in their free time when they can. try to remember [...]

I apologize if I seemed inconvenient or demanding, it was not my intention, I had not read that it was a project done just as a hobby, if I had read that this is not done continuously from the beginning I would not have been as demanding as I probably was, I apologize again for the inconvenience.

nononymousse, 2024-04-08:
i think you can assume that any free/open source project is someone(s) doing it as a hobby on their free time (even the big names, i invite you to check the xkcd 2347 comic about it), and i think it’s especially true in hackintosh. it’s mostly random peoples wanting to make things work in their free time

binxr, 2024-04-08:
> i think you can assume that any free/open source project is someone(s) doing it as a hobby on their free time (even the big names, i invite you to check the xkcd 2347 comic about it), and i think [...]

So I apologize, I was unnecessarily rude, I didn't know how to deal with this frustration and I ended up demanding too much from the developer. I hope you can understand that I was frustrated that a project that had been going on for so long wasn't working as it should for me.

binxr, 2024-04-08:
I really felt bad, I might be pushing myself too hard, the dev doesn't even care about my existence

VisualEhrmanntraut, 2024-04-08:
Guys, calm down.