ChefKissInc/NootedRed #246: Graphics AMD radeon graphic 680m

vishu002, 2024-04-07:
### macOS Version

Sonoma

### What is your CPU's model?

AMD RYZEN 6800hs

### Please describe the behaviour in detail.

when i insert Nred kexts file then after restart it can show amd radeon graphic 14mb, idk why showing low megabyte of iGPU , how to fix this problem ?

### What should've happened instead?

-

### If applicable, attach the .gpuRestart, .panic, etc file related to this issue.

-

### Bug report submission terms

- [X] I am using the latest version of NootedRed.
- [X] I am on the latest minor and patch version of macOS (ex. 7 is the minor and 10 is the patch of macOS 11.7.10).
- [X] The bug only happens with NootedRed enabled.
- [X] I have ruled out any external factors that could interfere, including, but not limited to, using OCLP or WhateverGreen.

VisualEhrmanntraut, 2024-04-07:
This is an RDNA iGPU, it is not supported yet. See #92

closed 2024-04-07 by VisualEhrmanntraut

vishu002, 2024-04-07:
how to solve this problem any idea?

VisualEhrmanntraut, 2024-04-07:
Sir, there's no solution, it's plain not supported; you have to wait a while.

vishu002, 2024-04-07:
OK sir ,i know sir this is more complicate to fix this problem ; can you pls tell me how much time to fixed this problem?