ChefKissInc/NootedRed #247: Artifacts on almost every app downloaded

binxr, 2024-04-08:
### macOS Version

Catalina

### What is your CPU's model?

AMD Ryzen 5 3350G PRO

### Please describe the behaviour in detail.

When I try to open Chrome or Firefox or Spotify or League of Legends itself, very heavy artifacts appear in which I can no longer use the system without restarting the Hackintosh I can't open these applications And I use macOS Monterey version 12.7.4, anyone Would you know how these artifacts could disappear?
[image]
[image]

### What should've happened instead?

Applications should appear normally without artifacts As is happening now

### Bug report submission terms

- [X] I am using the latest version of NootedRed.
- [X] I am on the latest minor and patch version of macOS (ex. 7 is the minor and 10 is the patch of macOS 11.7.10).
- [X] The bug only happens with NootedRed enabled.
- [X] I have ruled out any external factors that could interfere, including, but not limited to, using OCLP or WhateverGreen.

VisualEhrmanntraut, 2024-04-08:
These are all OpenGL apps. The issues tab are not a support forum for you to post your every question on.

VisualEhrmanntraut, 2024-04-08:
Duplicate of #158

closed 2024-04-08 by VisualEhrmanntraut