ChefKissInc/NootedRed #248: About HDMI Audio Control and Display Brightness control

Sayor00, 2024-04-09:
### macOS Version

Sonoma

### What is your CPU's model?

AMD Ryzen 5 5600G

### Please describe the behaviour in detail.

The HDMI audio controller is greyed out in the Audio settings and the Monitor's Brightness control is also grayed out
[image]
[image]

### What should've happened instead?

Instead of being greyed out, I should be able to control the brightness and audio

### Bug report submission terms

- [X] I am using the latest version of NootedRed.
- [X] I am on the latest minor and patch version of macOS (ex. 7 is the minor and 10 is the patch of macOS 11.7.10).
- [X] The bug only happens with NootedRed enabled.
- [X] I have ruled out any external factors that could interfere, including, but not limited to, using OCLP or WhateverGreen.

VisualEhrmanntraut, 2024-04-09:
HDMI audio is #225, brightness not working on Renoir means EFI misconfigured, see https://chefkissinc.github.io/nred#prerequisites

Sayor00, 2024-04-09:
> HDMI audio is #225, brightness not working on Renoir means EFI misconfigured, see https://chefkissinc.github.io/nred#prerequisites

What should I do to enable brightness control?

VisualEhrmanntraut, 2024-04-09:
Add the files the page tells you to add

VisualEhrmanntraut, 2024-04-09:
I linked you to the section right above the backlight one by accident, I edited the comment just now.

Sayor00, 2024-04-09:
> I linked you to the section right above the backlight one by accident, I edited the comment just now.

It's a desktop PC and an external monitor. Should I use those?

VisualEhrmanntraut, 2024-04-09:
I just noticed you have a desktop, sorry. You can't control the brightness on an external display, that's not how macOS is designed. You can use an external app such as MonitorControl.

Sayor00, 2024-04-09:
> I just noticed you have a desktop, sorry. You can't control the brightness on an external display, that's not how macOS is designed. You can use an external app such as MonitorControl.

Ohhhh. BTW Thanks. But but my CPU is Cezzanne. HDMI audio is coming out but I can't control the volume. its grayed out

VisualEhrmanntraut, 2024-04-09:
Cezanne is a Renoir derivative, yes. I assumed you had no audio at all, HDMI audio control is not allowed by macOS, again. You will have to use an external app or control the audio volume using the display settings.

Sayor00, 2024-04-09:
> Cezanne is a Renoir derivative, yes. I assumed you had no audio at all, HDMI audio control is not allowed by macOS, again. You will have to use an external app or control the audio volume using the display settings.

I used monitor control app. I can adjust brightness but I can't adjust HDMI audio

VisualEhrmanntraut, 2024-04-09:
I updated my comment to be more clear

Sayor00, 2024-04-09:
> I updated my comment to be more clear

Ok thanks a lot brother. But I used monitor control app. I can adjust brightness but I can't adjust HDMI audio. anything for this problem?

VisualEhrmanntraut, 2024-04-09:
I can't recall the name of the app, sorry. I don't think you can do this.

Sayor00, 2024-04-09:
I have a suggestion... Can you upload the updated kexts in the Releases section like others? It's just a suggestion..

VisualEhrmanntraut, 2024-04-09:
That's unrelated to this issue thread, but the reason my kexts like NootedRed and NootRX don't have releases yet is because they're not 100% stable. A release will be made once they're stable enough.

Sayor00, 2024-04-09:
And one more thing. Side topic but. What is the sound control key of the keyboard for macOS . My external keyboard has just f1 to f12 keys. and no fn toggle key. This is the last question. I wnt bother you again :(

VisualEhrmanntraut, 2024-04-09:
As far as I know, macOS only supports the key codes behind the non-fn key actions (when the key doesn't trigger F<n> but instead an action like volume up or down). There might be a way to change it in the settings, an app, or a key bind I'm missing.

Sayor00, 2024-04-09:
I have F1 to F12 keys. None of those are controlling volume. Do you know the shortcut? 🙁 and there is nothing in Keyboard shortcut in settings panel

Sayor00, 2024-04-09:
Okay have e good day. Bye 🙁

VisualEhrmanntraut, 2024-04-09:
Sir, this is just a hobby project, I don't constantly look at here.

Sayor00, 2024-04-09:
> Sir, this is just a hobby project, I don't constantly look at here.

ohhhh. But thanks for this kexts. Without your kexts, we would never get AMD iGPU support on macOS