ChefKissInc/NootedRed #250: PLZ HELP, BLACK SCREEN with NootedRed, tried all macOSversions from Big Sur to Venture, still same result

medicupstairs, 2024-04-13:
### macOS Version

Big Sur

### What is your CPU's model?

AMD Ryzen 5 4600H

### Please describe the behaviour in detail.

black screen after verbose..

### What should've happened instead?

Apple recovery mode should've loaded

### Bug report submission terms

- [X] I am using the latest version of NootedRed.
- [X] I am on the latest minor and patch version of macOS (ex. 7 is the minor and 10 is the patch of macOS 11.7.10).
- [X] The bug only happens with NootedRed enabled.
- [X] I have ruled out any external factors that could interfere, including, but not limited to, using OCLP or WhateverGreen.

medicupstairs, 2024-04-13:
Okay so I got to the recovery
but it is helllll slowww (no acceleration)
during verbose, I don't see NootedRed in the loading list
my device is Lenovo Ideapad Gaming 3
RYZEN 5 4600H , disables GTX 1650

VisualEhrmanntraut, 2024-04-13:
Use the Telegram group for support. (Linked in our site)

closed 2024-04-13 by VisualEhrmanntraut

medicupstairs, 2024-04-13:
Okayyy I'll se thanks 🫡