ChefKissInc/NootedRed #251: i have ryzen 5 7535hs with amd radeon 660m what kind of time to add igpu of ryzen radeon gpu ??

Mohammedkarms, 2024-04-17:
### macOS Version

Ventura

### What is your CPU's model?

 ryzen 5 7535hs

### Please describe the feature in detail.

i want nootedRed to add support to my apu ryzen 5 7535hs  or rdna igpu please help me  and what kind of time to add  it 
and thanks for all the efforts 

### What alternatives have you tried, if any?

i test all the method to install mac but it is bad experience 
[image]

### Feature request submission terms

- [X] I am using the latest version of NootedRed, thereby extension the feature is not already present.
- [X] I am on the latest minor and patch version of macOS (ex. 7 is the minor and 10 is the patch of macOS 11.7.10).
- [X] The feature being present in NootedRed makes sense.
- [X] I have ruled out any external factors that could interfere, including, but not limited to, using OCLP or WhateverGreen.

VisualEhrmanntraut, 2024-04-19:
Sir, this is a hobby; we do not have specific time slices. also this is not a feature request, you are just asking for an ETA.

closed 2024-04-19 by VisualEhrmanntraut

lsslsslss, 2024-06-16:
Same CPU Same iGPU. Whether Amd radeon 660m can be add to new NootedRed Version?