ChefKissInc/NootedRed #255: Internal Display Become Blurred and Flickering

BreezeConfirmingWms, 2024-04-27:
### macOS Version

Ventura

### What is your CPU's model?

AMD Ryzen 9 4900HS with Radeon Graphics

### Please describe the behaviour in detail.

My laptop is ASUS ROG Zephyrus-G15-GA502IV,and its internal display not work properly,the color is grey and the display become blurred and flickering

when I connect to my external 2k display, the internal display is broken grey:
[image]
when I disconnect and use internal display alone
[image]

the infomation of internal monitor shown in terminal is:
 ```wiki
 LQ156M1JW09:
      Resolution: 1920 x 1080 (1080p FHD - Full High Definition)
      UI Looks like: 1920 x 1080 @ 240.00Hz
      Framebuffer Depth: 30-Bit Color (ARGB2101010)
      Mirror: Off
      Online: Yes
      Automatically Adjust Brightness: Yes
      Connection Type: Internal
```
Additionaly,I found lower refresh rate like transfer 240hz to 60hz could help erase the black stripes in the picture above while there is still a blurred and grey screen probelm 

### What should've happened instead?

The built-in monitor should display normally

### If applicable, attach the .gpuRestart, .panic, etc file related to this issue.

I just command `/System/Library/Extensions/AppleGraphicsControl.kext/Contents/MacOS/AGDCDiagnose -a`  to get:
[agdcdiganose.txt](https://github.com/ChefKissInc/NootedRed/files/15138260/agdcdiganose.txt)

VisualEhrmanntraut, 2024-05-01:
The build of this kext in the system is ancient. LOL @ u.
[image]

VisualEhrmanntraut, 2024-05-01:
oh... wait, I think the patch that changes the value is actually broken 🤦🏻‍♂️ My bad

VisualEhrmanntraut, 2024-05-01:
anyway... the problem is that macOS detected your laptop's pixel encoding as YCbCr444

VisualEhrmanntraut, 2024-05-02:
You can try creating an override file using https://gist.github.com/adaugherity/7435890
the files created with it should go in `/Library/Displays/Contents/Resources/Overrides`

BreezeConfirmingWms, 2024-05-02:
Thanks for your reply. I have tried to override file but my internal display problem still exists.And I check my `AGDCDiganose -a` info,it shows my internal dispaly pixel encoding is **RGB444**. The previous txt  uploaded I can find `OUTPUT: Pixel Encoding	1 (RGB444)` for my internal display `Window (Active)	1920 x 1080`,and YCbCr444 for My external display `Window (Active)	2560 x 1440` ,which is connected through HDMI port.So I think it no matter. When i use internal display alone, the grey screen doesn't change at all.

I'm compeletly an rookie of macos or hackintosh, so could you give me more debug command to check the configuration or other useful logfile?
If it's outside of the scope of support ,I feel regretful about this. Thanks again!I will continue to pay attention to your guidance until the problem is closed

BreezeConfirmingWms, 2024-05-02:
Below is output of `ioreg -l -d0 -w 0 -r -c AppleDisplay`
```wiki
+-o AppleDisplay  <class AppleDisplay, id 0x100000662, registered, matched, active, busy 0 (0 ms), retain 9>
    {
      "IOClass" = "AppleDisplay"
      "CFBundleIdentifier" = "com.apple.iokit.IOGraphicsFamily"
      "IOProviderClass" = "IODisplayConnect"
      "DisplayProductID" = 10021
      "IODisplayEDID" = <00ffffffffffff004d672527010000002b210103803c2178af7c56ab5048a7270b5053a7cf0081c081809500a9c0b300d1c0010101015d5e00a0a0a029503020350055502100001e023a801871382d40582c450055502100001e000000fd00309219fa3c000a202020202020000000fc00483237543232530a202020202001d3020347f34f0102030405101213141f5d5f61603f23090707830100006a030c001000387820000067d85dc40178c003681a000001013092ede305c000e60607015d4937e30f003c197600a0a0a02a503040450055502100001e6fc200a0a0a055503020350055502100001ef9e300a0a0a032501040350055502100001e00006a>
      "IODisplayPrefsKeyOld" = "IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/GP17@8,1/IOPP/IGPU@0/AMDRadeonX6000_AmdRadeonControllerNavi10/ATY,RadeonFramebuffer@1/display0/AppleDisplay-4d67-2725"
      "IODisplayPrefsKey" = "IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/GP17@8,1/IOPP/IGPU@0-4d67-2725-e0edc1fca52fb1cc"
      "IOProbeScore" = 2000
      "IOPowerManagement" = {"CapabilityFlags"=49152,"MaxPowerState"=3,"CurrentPowerState"=3}
      "IODisplayParameters" = {"pscn"={"min"=7500,"max"=10000,"value"=10000},"rgsc"={"min"=0,"max"=65536,"value"=65536},"bgsc"={"min"=0,"max"=65536,"value"=65536},"vblm"={"min"=0,"max"=196608,"value"=65536},"ownr"=4294968542,"ggsc"={"min"=0,"max"=65536,"value"=65536}}
      "IOMatchCategory" = "IODefaultMatchCategory"
      "IOPersonalityPublisher" = "com.apple.iokit.IOGraphicsFamily"
      "IODisplayConnectFlags" = <00000000>
      "CFBundleIdentifierKernel" = "com.apple.iokit.IOGraphicsFamily"
      "IODisplayAttributes" = {"IODisplayVSDB1A0000"=<01013092ed>,"IODisplayVSDB030C00"=<10003878200000>,"IODisplayAttributes"=<676174760000000073676c6600000000726c6f63110100006364670080808000676c666401000000>,"IODisplayVSDBD85DC4"=<0178c003>}
      "DisplayVendorID" = 19815
      "DisplaySerialNumber" = 1
    }
```

VisualEhrmanntraut, 2024-05-02:
honestly, the driver stack for this part of macOS is designed pretty badly

MinecraftNerrd1337, 2025-01-22:
Could there be a world where this issue is just due to the display being eDP 240Hz, and maybe even happens to all of them?

VisualEhrmanntraut, 2025-03-22:
Is this still happening with latest commit?

MinecraftNerrd1337, 2025-04-14:
Sorry for not seeing your reply earlier. I'll be able to try in about a week, as I'll have that 240Hz display to test again.

MinecraftNerrd1337, 2025-04-25:
> Is this still happening with latest commit?

Can confirm it is. The issue isn't fixed at all

VisualEhrmanntraut, 2025-11-16:
is it fixed now

closed 2025-11-28 by VisualEhrmanntraut