ChefKissInc/NootedRed #258: Menubar graphical glitch

Svetrikk, 2024-04-29:
### macOS Version

Sonoma

### What is your CPU's model?

Ryzen 5 4500U

### Please describe the behaviour in detail.

Sometimes the menubar gets colored. It usually happens when I open an app or something. The background of a button in a menubar gets only colored but disappears once I click on the button. The whole bar gets colored sometimes but also disappears when I open something in the menubar (eg. control center). Also this wasn't happening on Ventura but is on Sonoma.

[image]

### What should've happened instead?

The menubar should be transparent

TahsinFaiyaz30, 2024-04-30:
This happens because of insufficient VRAM. increase your VRAM to at least 2GB

VisualEhrmanntraut, 2024-05-29:
Timed out waiting for response.

closed 2024-05-29 by VisualEhrmanntraut