ChefKissInc/NootedRed #260: Display not detected over HDMI connection.

darthsian, 2024-05-02:
### macOS Version

Ventura

### What is your CPU's model?

AMD Ryzen 7 5700U

### Please describe the behaviour in detail.

With the last commit (da6376b), the monitor connected via HDMI is not working. DisplayPort works fine.

### What should've happened instead?

The monitor via HDMI should work.

darthsian, 2024-05-02:
The last commit where HDMI works without problems is 0c15616.

VisualEhrmanntraut, 2024-05-02:
```sh
sudo rm /var/db/.com.apple.iokit.graphics
```
see if the issue persists, if yes, provide an AGDCDiagnose log using 
```sh
sudo /System/Library/Extensions/AppleGraphicsControl.kext/Contents/MacOS/AGDCDiagnose
```
and send it over here

VisualEhrmanntraut, 2024-05-02:
AGDP printed no frame buffers which means it does not like your configuration, switch to a different SMBIOS

darthsian, 2024-05-02:
I use iMac20,1 on all AMD Mini PCs and there has been no problem so far. Why has it worked without problems until now?

VisualEhrmanntraut, 2024-05-02:
The getVendorInfo patch was broken until now, making AGDC think this is a dedicated GPU.

darthsian, 2024-05-02:
Does it help if I send you the AGDCDiagnose log with commit 0c15616 where it still works?

darthsian, 2024-05-02:
> The getVendorInfo patch was broken until now, making AGDC think this is a dedicated GPU.

Hmm, so what SMBIOS should we be using now if not iMac20,1? The iMacPro1,1 only has only dedicated GPU...

VisualEhrmanntraut, 2024-05-02:
Try `MacBookPro16,3`.

darthsian, 2024-05-02:
> Try `MacBookPro16,3`.

So, it doesn't strike me as a good idea to use a SMBIOS of laptop for a desktop.

VisualEhrmanntraut, 2024-05-02:
mini PCs don't really strike me as AIOs either :^)

techmare, 2024-05-02:
bruh ur mini pc literally have a laptop cpu 💀💀

techmare, 2024-05-02:
literally a laptop motherboard  shrunken in size

darthsian, 2024-05-02:
> Try `MacBookPro16,3`.

Ok, so on the `MacBookPro16,3` HDMI also unfortunately does not work :-/

VisualEhrmanntraut, 2024-05-02:
That's weird.

darthsian, 2024-05-02:
Isn't this a similar problem to the Intel Hackintosh where you have to use agdpmod=pikera boot argument with newer GPUs, otherwise the computer boots to a black screen?

darthsian, 2024-05-02:
> this patch is already in place, along with another for making AGDP continue if there's only one FB

Look like we need some new agdpmod patch...

SalmonPete, 2024-05-02:
I'm experiencing the same issue over HDMI on my 4700G Desktop, 14.4.1, iMac20,2, commit 2cf2ad6.

VisualEhrmanntraut, 2024-05-03:
I have removed the patch at fault. Sorry for the inconvenience, I hate AGDP.

SalmonPete, 2024-05-04:
Commit [e450df1] is still not detecting HDMI. [0c15616] is working great.

VisualEhrmanntraut, 2024-05-04:
what

VisualEhrmanntraut, 2024-05-05:
I reverted the problematic commit, and will reintroduce the fix properly once I fully understand what I'm supposed to do

closed 2024-05-05 by VisualEhrmanntraut

darthsian, 2024-08-11:
With the latest version I have the same problem again - HDMI does not work.

VisualEhrmanntraut, 2024-08-11:
There have been no code changes for 2 months. That's impossible.

darthsian, 2024-08-11:
So I reverted back to the last version again and now it works. I tried it several times before and it didn't work. I don't understand. Sorry for the false alarm.