ChefKissInc/NootedRed #265: Constant system crashed (usually caused by apple apps)

BeluwuGANG, 2024-05-17:
### macOS Version

Sonoma

### What is your CPU's model?

AMD Ryzen 5 3500U

### Please describe the behaviour in detail.

When you open ex. Apple Podcasts, the system crashes and then reboots

### What should've happened instead?

The podcasts app should've ran like in Ventura

### If applicable, attach the .gpuRestart, .panic, etc file related to this issue.

This is NootedRed problem, no need for logs.

yarecco, 2024-05-18:
Can't confirm. Podcasts App works without any issues on Sonoma 14.5 with NootedRed.
[image]

VisualEhrmanntraut, 2024-05-29:
Duplicate of #235

closed 2024-05-29 by VisualEhrmanntraut